morrownr/8812au-20210820 #16: ALFA AWUS036ACH not working in USB3.0 mode on 2.4Ghz networks

dill-shower, 2024-03-29:
Brief description of the problem: when setting the config to use usb3, the adapter connects to 2.4Ggz networks, but the connection speed is very slow (30kb/sec).
This problem does not appear in
A) Past driver version
B) On 5Ghz netwoks
C) If I delete thin config parametr and use usb2

P.s: setting the region in config also doesn't work. I set to use US region, iw reg get command shows that region 00 UNSET. Self region setting via iw reg set works.
I dont use VM. It's host OS Debian bookworm stable 

```

Installation logs:

: ---------------------------
: install-driver.sh v20240314
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 65245980 (total system memory)
: 6.1.0-18-amd64 (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/lkrg/0.9.7/source/dkms.conf)
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/lkrg/0.9.7/source/dkms.conf)
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
kernelver=6.1.0-18-amd64 ./dkms-make.sh.........
Signing module /var/lib/dkms/rtl8812au/5.13.6-23/build/8812au.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8812au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-18-amd64/updates/dkms/
depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates to this driver SHOULD be performed before major
[... 281 lines cut ...]
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: 8812A: EDCA_BE_UL=0x5ea42b EDCA_BE_DL =0x5ea42b
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)
 RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
 RTW: 8812A: EDCA_BE_UL=0x5ea42b EDCA_BE_DL =0x5ea42b
```

morrownr, 2024-03-30:
Hi @dill-shower 

I'll put this on my to-do list.

lwfinger, 2024-03-30:
My AWUS036 only has a USB2 interface. In the USB connector, what color is the spacer tab?

dill-shower, 2024-03-30:
> My AWUS036 only has a USB2 interface. In the USB connector, what color is the spacer tab?

Hi. Blue. It's awus036ACH model. 
[image]

morrownr, 2024-03-31:
> If I delete the config parameter rtw_switch_usb_mode=1

Since usb3 is of no use with the 2.4 GHz band, select 0 or leave the parameter out. In fact, I am pretty sure that is what I recommend in my AP Mode guide:

The Main Menu for this site is:  https://github.com/morrownr/USB-WiFi

The AP Mode guide is menu item 8.

lwfinger, 2024-03-31:
@dill-shower - One more test is to run 'lsusb -v | less' and scroll down to the entry for the AWUS036 entry and check the value of bcdUSB.

dill-shower, 2024-03-31:
> @dill-shower - One more test is to run 'lsusb -v | less' and scroll down to the entry for the AWUS036 entry and check the value of bcdUSB.

3.00 if I use rtw_switch_usb_mode=1

morrownr, 2024-03-31:
> P.s: setting the region in config also doesn't work.

The region only matters if you are running AP mode. Are you running AP mode? In fact, you probably don't need to set the country code unless you are running AP mode and maybe not then.

> Deprecated feature: REMAKE_INITRD (/var/lib/dkms/lkrg/0.9.7/source/dkms.conf)

Have you installed something that could account for the above?

> RTW: unknown vendor specific information element ignored (vendor OUI 00:e0:4c len=6)

Not sure what is going on there. I'll have to take a look.

dill-shower, 2024-03-31:
> Have you installed something that could account for the above?

Yes. It's LKRG module. This should have nothing to do with rtl8812au.

morrownr, 2024-03-31:
It should have nothing to do with rtl8812au but it is showing up in the onscreen log from the script. Can you show me?

$ dkms status

dill-shower, 2024-04-01:
> $ dkms status

Deprecated feature: REMAKE_INITRD (/var/lib/dkms/lkrg/0.9.7/source/dkms.conf)
lkrg/0.9.7, 6.1.0-18-amd64, x86_64: installed
rtl8812au/5.13.6-23, 6.1.0-18-amd64, x86_64: installed

dill-shower, 2024-04-02:
> @dill-shower - Please extract the section on the AWUS036 that comes from 'lsusb -v'. I want to see if there is a definitive variable that will let us set USB 3 in the drivers.

Now without usb3 flag output is this. If need with usb3 flag please write

Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812 RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

lwfinger, 2024-04-02:
That post is sufficient. I had hoped to see something that indicated the device was USB 3, but running as USB 2. So far, nothing seems to be appropriate. It does not appear that we have any way to tell. I am looking into the code  that does the change to see what it does and why it is not auto-adjusting. There is one setting in the efuse that would prevent it from changing, but that seems not to apply to your system.