morrownr/8812au-20210820 #17: raspberry pi 5 os bookworm?

LAP47, 2024-04-24:
Getting a "Your kernel header files aren't properly installed. Please consult your distro documentation or user support forums. Once the header is properly installed, please run "sudo ./install-driver.sh" Anyway to get this working on there so that I can use it to make a portable router? I'm guessing that it's not for aarch64.

lwfinger, 2024-04-24:
It should work for aarch64. Please provide the complete output from 'make'.

LAP47, 2024-04-24:
How do I do that?

lwfinger, 2024-04-24:
If you are answering my question, copy and paste from the terminal to the web browser.

If you mean FAQ.md, 'less FAQ.md'.

morrownr, 2024-04-24:
> If you mean FAQ.md, 'less FAQ.md'.

Or he can call up this repo in a web browser and click on the file FAQ.md.

LAP47, 2024-04-24:
How do I provide the complete output from make?

LAP47, 2024-04-24:
I'm getting the Your kernel header files aren't properly installed. Please
consult your distro documentation or user support forums.
Once the header files are properly installed, please run ...... On a 64bit
installation of raspberry pi os. Even the driver that's available on Ubuntu
23.10 repo's doesn't work. Meaning that it installs fine via synaptic but
doesn't make the dongle work. No lights come on it.

morrownr, 2024-04-24:
@LAP47 

> How do I provide the complete output from make?

From the driver directory:

$ make

copy the screen output and post it.

LAP47, 2024-04-24:
I am answering you and there is no make command in the process that I'm
using. It's just

$sudo apt update

$sudo apt install git dkms

$git clone https://github.com/morrownr/8812au-20210820.git

$cd 8812au-20210820

$sudo ./install-driver.sh

morrownr, 2024-04-24:
@LAP47 

> I am answering you and there is no make command in the process that I'm
using. It's just...

You are following the primary installation instructions. That is fine but @lwfinger wants to see the results as if you were doing a manual installation. The manual installation is also available in the README but all you need to do is:

From the driver directory, run:

$ make

copy the output and post it here.

This output could provide more details than the installation script provides when it comes to things like header files.

LAP47, 2024-04-24:
> Dead adapter? Dead USB port?
> Are you seeing it when you run:
> $ lsusb
> For what it is worth, I cloned this repo to my RasPi4B last night. It runs
> as a dual band AP with relatively clean 64 bit installation of the
> 2024-03-15 RasPiOS. It worked. This message is going through it. I did not
> see any issues at all.

Yes. It works on Linux Mint Debian 6 but not on the raspberry pi 5 and
thank you for keeping this thing alive. Not on Raspberry Pi OS , Ubuntu
23.10, or MX Linux Respin for the Pi.

michelwi, 2024-04-25:
You did install the headers, right?
```
sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git
```
And there is also not a kernel update pending and you did not reboot in a wee while?

LAP47, 2024-04-25:
I did not install anything else because I didn't know that I had to. And I
don't know about the second part. I don't know what you mean there about
pending kernel updates and rebooting in a wee while. Lol I'll install the
headers and whatever else you want me to do in whatever order you want me
to do it just tell me what to do and I'll do it because I was only going by
a quick Google search that I did and who knows if that person or persons
even knew what they were doing but like I said I followed those quick
instructions with Linux Mint Debian 6 and it worked without having to do
anything additional.

michelwi, 2024-04-25:
> just tell me what to do and I'll do it 

Fortunately we have very good step-by-step instructions in the readme file, which you can read online
https://github.com/morrownr/8812au-20210820?tab=readme-ov-file#installation-steps
or by opening the cloned repositories README file on your disk. (Search for installation steps)

The command I posted is part of the instructions (Step 3).

> I don't know what you mean there about pending kernel updates and rebooting

Sometimes when you do software updates on the raspberry there will be a new kernel installed. This is one of the few times when one needs to restart the computer (similar to Windows updates). If you then do not restart manually, there can be issues where the os thinks the new version was valid, but actually you are still using the old version.

If you are unsure and don't mind, I would try rebooting just in case. A reboot is also part of the step-by-step guide linked above.

LAP47, 2024-04-25:
Okay. Thanks and sorry for the trouble.

michelwi, 2024-04-25:
No worries. Please let us know if this solves your problem or if you need further assistance.

LAP47, 2024-04-25:
Is there a way that I can print it out? or do I have to click and drag on
it all to do that?

michelwi, 2024-04-25:
Sorry I do not understand the question. 
Are you asking how to print the README file on a sheet of paper?

I would recommend opening the link in a browser window on the device that you are using to interact with the raspberry and then copy pasting the appropriate lines in the terminal

LAP47, 2024-04-25:
Yes. I like being able to hold it in my hand and read it that way instead
of having to look at on the screen and scroll up and down. Lol Thanks. I'll
let you know how it goes.

michelwi, 2024-04-25:
Sure you can physically print it. Open the readme file in a browser of your choice 
https://github.com/morrownr/8812au-20210820/blob/main/README.md
Then most browsers should open the print dialog when pressing [ctrl]+p

Exact steps of cause depend on your operating system, browser, printer.

But I still recommend working in a browser and scrolling up and down (maybe opening the same tab multiple times?) because if you are typing the commands by hand off of a sheet of paper, there might be an increased risk of typoing compared to copy pasting the whole line.

Good luck.

LAP47, 2024-04-25:
Okay. Thanks.

LAP47, 2024-04-25:
Didn't work. Still coming up with the same error message after sudo
./install-driver.sh. Is the kernel too new maybe? Should I not upgrade the
kernel at all via rpi-update?

alkisg, 2024-04-25:
rpi-update isn't supported by the Raspberry Pi OS developers, they do not provide the kernel headers for that.
You need to revert to the official kernel by following these instructions:

https://raspberrypi.stackexchange.com/questions/101790/how-to-revert-from-rpi-update-to-stable-build

I.e. open a terminal and run these commands:

    sudo apt-get update
    sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel

michelwi, 2024-04-25:
Okay, then lets walk you through the steps that morrownr and lwfinger asked.

> Take a look in the file FAQ.md. The last item very well may apply to your situation.

The last point of the FAQ https://github.com/morrownr/8812au-20210820/blob/main/FAQ.md mentions running a 32 bit version of the operating system with a 64 bit kernel.
To verify if this is the case run
```
uname -m
```
It should show one out of 
```
32-bit: armv6l armv7l armv8 armhf
64-bit: arm64 aarch64
```
This is the version of your kernel. To find the version of the operating system, please run
```
getconf LONG_BIT
```
It should print either 32 or 64.

If kernel is 64 and os is 32, then we know how to fix the problem. Otherwise proceed to the next comment:

> The manual installation is also available in the README but all you need to do is:
>
> From the driver directory, run:
>
> ```$ make```
>
> copy the output and post it here.

Where "the driver directory" is the exact same folder where you run the automatic installation script (`$sudo ./install-driver.sh`) in the last step.

If you need help copying and posting the output of make, then please let us know how you operate your raspberry pi (e.g. via ssh acces from a different pc, or with mouse/keyboard/monitor directly attached) so we can try to best guide you.

morrownr, 2024-04-25:
@LAP47 

I think I have figured out what is going on based on one of your earlier post.

> I am answering you and there is no make command in the process that I'm
using. It's just...

The steps you posted are wrong for what you are doing. What you need to do is find the `Installation Steps` section in the README file and follow the steps very carefully. Some of the steps are different depending on the Linux distro you are using. You can ask questions on each step if you need to do so.

We can't do this for you. A lot of work has gone into the documentation. It attempts to make it easy on those who are not familiar with the internal workings of operating systems.

If this is something you are not comfortable doing, then my advice is to get a USB WiFi adapter that will be plug and play so that all you have to do is plug it in and go. The Main Menu to this site is located:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 should be helpful. Menu item 2 shows a long list of USB WiFi adapters that are know to work well with Linux and require no driver compiling. You plug the adapter in and go.

LAP47, 2024-04-25:
No. I did follow those instructions last night and kept running into the
same thing but it's probably because I was upgrading the kernel via
rpi-update along with the update and upgrading process even though it says
not to because that's how I roll I guess but now I'm running into another
problem with raspberry pi os giving me trouble with upgrading after a clean
install. The software wolfram-engine doesn't want to upgrade and even after
uninstalling it and trying to re-install it it doesn't want to install
properly. This is why I wish that there a proper version of Debian for it
but there isn't one yet for the raspberry pi 5. It gets stuck or takes
forever at the unpacking process and then ends up failing altogether.

LAP47, 2024-04-25:
It's working now in raspberry pi os but I don't like raspberry pi os so I'm
going to try it in the lite version and in Ubuntu 23.10 desktop and server
edition and see how it goes. Thanks.

LAP47, 2024-04-26:
It's not working on MX23.2_rpi_respin_arm64, not working on Ubuntu 23.10
with kernel 6.5.0-1015-raspi either. Both are giving me that same Tour
kernel header files aren't properly installed error even though I installed
them and saw no errors there.

morrownr, 2024-04-26:
For your Ubuntu 23.10 system:

Go to the driver directory and run:

sudo apt install -y build-essential dkms git iw

sudo sh install-driver.sh

reboot when the script asks.

FYI: I have a test system that has Ubuntu 23.10 on it. I have tested. This driver installs and works fine. This repo gets over 100 hits per day as I have not taken the older version of the driver down yet. It will get up to 200 hits per day soon. My point here being that this is a heavily used driver and I'm not seeing this from any other user so something specific and unusual is going on.

LAP47, 2024-04-26:
I did that. I followed the installation steps. Without updating the kernel
unless it does it automatically when apt updating and apt upgrading and
this is on the raspberry pi 5 8G. I'm in the process of reimaging it via
NETBOOT because when I did it the first time from my laptop and then booted
into and tried updating it and upgrading it was giving me trouble. Even the
setup process was giving me trouble for some reason. I had to shut it off
with the power button and then restart it and do the setup process all
over again. But even the driver that's available in the driver's manager
installed but didn't get the dongle to work. So idk what's going on but
I've gotten yours to work on my HP ProBook 6570b with Linux Mint Debian 6
and on the raspberry pi 5 8G with raspbian OS.

LAP47, 2024-04-26:
I'll try that first thing that you suggested [sudo apt install -y
build-essential dkms git iw ] when I get chance because I wasn't doing that
from the driver directory. I was doing that after updating and upgrading I
think or whatever order the installation steps was telling me to do which
was after the reboot from updating and upgrading.

LAP47, 2024-04-26:
Nope. Still not working on Ubuntu 23.10 on the raspberry pi 5. Won't
install.

morrownr, 2024-04-26:
>  Still not working on Ubuntu 23.10 on the raspberry pi 5. Won't install.

I don't test the RasPi version of Ubuntu but was hoping the regular Ubuntu steps would work. Your best bet might be to ask in an appropriate RasPi forum. What you need to know is how to install the kernel header files.

michelwi, 2024-04-26:
Just out of curiosity, the raspberry pi 5 does have built-in wifi. If this stick is giving so much problems, would it be feasible to use built-in wifi instead?

LAP47, 2024-04-26:
Okay thanks.

morrownr, 2024-04-26:
@michelwi 

> ... the raspberry pi 5 does have built-in wifi. 

It uses a Broadcom chip. Broadcom's Linux support is subpar. This is one of the reasons you see so many Pi users using USB WiFi adapters. The best solution for Linux users that just want an adapter that just works is to look at the Plug and Play List on the Main Menu.

LAP47, 2024-04-29:
I did it. I got it working on the raspberry pi 5 in Ubuntu 24.04 LTS.

LAP47, 2024-04-29:
Then why would the raspberry pi foundation choose them (Broadcom chips) to
use then?

morrownr, 2024-04-29:
@LAP47 

> I did it. I got it working on the raspberry pi 5 in Ubuntu 24.04 LTS.

Did you do it with the installation steps in the README or was something different required? I ask because I can add to the README to clarify things.

morrownr, 2024-04-29:
> Then why would the raspberry pi foundation choose them (Broadcom chips) to
use then?

I have heard that very early on former Broadcom employees were involved with RasPi and they had connections to Broadcom. Do I know this for a fact? No. What I do know that is factual is that Linux users should avoid Broadcom products if at all possible if they want good support.

Any startup that decides to make Linux part of their project needs to immediately determine the companies that provide the best Linux support for the products they will buy. Period.

LAP47, 2024-04-29:
Both the steps in the readme file and something else. I'm going to make a
text document of the steps that I took and send it to you but I have to
redo it because I tried something else and then ended up breaking the
system I think so I'll get back to you once I've done it again successfully
and made that document.

LAP47, 2024-04-29:
Right. I hear you there. Well hopefully they'll find someone else to work
with in the future and get rid of that chip.

morrownr, 2024-04-29:
> I'll get back to you once I've done it again successfully

Thanks. I try to make the drivers here work on as many distros as is feasible. You can imagine how difficult it is so help is needed.

Now you have an idea why I recommend adapters with in-kernel drivers.

LAP47, 2024-04-29:
How do you  even do that? What's the first step? And what do you do if and
when you get to the point that the driver won't work anymore. Here's the
text document I made with your steps and the ones that I've added that
worked for me. You can obviously edit it to your liking. I didn't know if I
was supposed to uninstall the old mismatched  kernel headers or not or even
how to do that from terminal so that's why I installed synaptic so I could
go ahead and do that but it terms out that it wasn't even necessary on a
clean install of ubuntu so the only thing that I used it for was to get rid
of the not installed residual configs that I guess were left over from the
setup process of ubuntu so it's really got nothing to do with the driver
installation steps. Do you know how to remove old or mismatched kernels in
the terminal? The only reason why I'm using this dongle is because it's the
newest one that I've got and it's 802.11ac and I want to use it to use the
raspberry pi as a portable hotspot which apparently ubuntu has built in now
so this will be connecting to the internet and then the raspberry pi's wifi
acts as the wireless hotspot. Otherwise you have to be connected to
ethernet and then use the pi's wireless as the hotspot.

LAP47, 2024-05-01:
I got it to install on MX Linux on the raspberry pi 5 too now but it's got
no lights and isn't working. I just uninstalled all of the older kernels
and then installed the headers again  and then I was able to download the
driver's

LAP47, 2024-05-01:
I just got it working on a fresh install of MX-Linux Respin for Raspberry
Pi 5,too. I think that it's because there were no matching headers last
night when I tried it but today there was. Kernel 6.6.28+rpt-rpi-2712 and
linux-headers-6.6.28+rpt-common-rpi
linux-headers-6.6.28+rpt-rpi-2712
Linux-headers-6.6.28+rpt-rpi-v8.

The same method that I used for raspberry pi os/Debian and using the
command - sudo apt install-headers-$(uname -r)

Thanks.

morrownr, 2024-05-01:
Glad to see you are making progress. Some distros like to play with the header files.

LAP47, 2024-05-01:
Yeah I don't know. I haven't looked at them myself and would have to look
at the original ones first I guess before making any judgements. I hope
that it helps you get more users or the drivers though.