morrownr/8812au-20210820 #21: Any ideas on how to set the ibss mode?

tallorate, 2024-05-27:
I've successfully installed the driver and it's okay running in AP and station mode. But when I trying to set the ibss mode, I can make two devices into the same cell.
`root@cly-virtual-machine:/home/cly# iw dev wlan0 link
Joined IBSS 02:11:87:86:a3:0f (on wlan0)
	SSID: myadhoc
	freq: 2437
root@cly-virtual-machine:/home/cly# iwconfig wlan0
wlan0     IEEE 802.11bg  ESSID:"myadhoc"  Nickname:"<WIFI@REALTEK>"
          Mode:Ad-Hoc  Frequency:2.437 GHz  Cell: 02:11:87:86:A3:0F   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=1/100  Signal level=1/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0`
 but there are no packet exchanges at all and I even couldn't ping through each other.

morrownr, 2024-05-27:
@tallorate 

I intended to mark ibss as untested but forgot to do so. All of the rest of the supported modes are tested.

To answer your question: I have no idea. I have never used that mode. If I had time, I would dig in and see what I could find out.

If you do figure it out, if it is working, let us know and I will post how to do it for those in the future.

tallorate, 2024-05-28:
Thanks to your reply. 
Another question is that do you have any idea about the wds and mesh which are mentioned in the `core` code, I've also tried to set the interface into these modes but unfortunately they failed.