morrownr/8812au-20210820 #22: Supported Driver

Asaf-E, 2024-05-27:
Hi,

I couldn't find the support list (it is only partial on the readme).

Does it support EW-7811UTC, or do I have to use a different driver?

In the past, this driver supported that chipset.

Thanks for the help.

morrownr, 2024-05-27:
You can check support by reading the file `supported-device-IDs` or you can paste the results of:

$ lsusb

...and I can help you.

morrownr, 2024-05-27:
I did some digging on that adapter. It uses the rtl8821au chip so this is not the correct driver, however you have some options.
The following repo supports your adapter:

https://github.com/morrownr/8821au-20210708

However, before you install it, let me give you the link to where we are working on a new mac80211 compliant driver that we intend to upstream to the kernel when it is done. We need testers. We are communicated in the following issue and the first message gives the link to the driver to test:

https://github.com/morrownr/8821au-20210708/issues/133

Regards

Asaf-E, 2024-05-28:
Hi, Thanks you very much for the help!
I'm not using a mac, but a linux ubuntu 22.04, I'd love to help out, but would it work? Thanks!

morrownr, 2024-05-28:
> linux ubuntu 22.04

That will work fine. I am talking about a Linux driver.

> mac80211

This is the name of one of the components in the Linux wifi support stack.

You can look at this thread:

morrownr/8821au-20210708#133

If you read it, you will see that we are coodinating work on the driver there. The first message gives you the link to where you can find the driver. Follow the instructions at that site. Let me know how it goes. The idea is for the driver to go into the Linux kernel that we don't have to look around for drivers.

Asaf-E, 2024-05-28:
Alright, I'll update.

Can I ask how to disable the second interface: p2p-dev-wlan0  wifi-p2p  disconnected  --
is this an option, or do I have to disable it manually? Where can I find the documentation on all the options?
 /etc/modprobe.d/8821au.conf
 contains only part of them, and ls /sys/module/8821au/parameters/ doesn't give information about what the parameters actually do

morrownr, 2024-05-28:
Can I get you to start new issue in the repo that is the right one for your adapter?

https://github.com/morrownr/8821au-20210708

I'll get lost if we work this here.

> Can I ask how to disable the second interface?

What hardware are you using?

Paste the results of:

$ iw dev

> Where can I find the documentation on all the options?

Heck... I had to go through the source to document the options that are documented. Did I mention that Realtek sucks and Mediatek is much better?

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 are good reads for Linux users that use usb wifi adapters and modules.

> Alright, I'll update.

The RTW88 new driver for your adapter is not an upgrade. It is a totally new driver. You have to uninstall what you have installed to install it.

Asaf-E, 2024-05-28:
Hi, Thanks for the help, I opened the issue under the other repo, 
https://github.com/morrownr/8821au-20210708/issues/138

Thanks you for the help, I'll close this thread

closed 2024-05-28 by Asaf-E