morrownr/8812au-20210820 #25: Monitor mode doesn't work (AWUS036ACH v2)

mentalpabulum, 2024-06-05:
What's interesting is that [this](https://github.com/aircrack-ng/rtl8812au) driver allows me to enable monitor mode, but I don't see any results (cells) when scanning. The previous version of your driver had no such problem.

$ lsusb:
`Bus 001 Device 006: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
`

$ rfkill list all:
```
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

$ dkms status:
`rtl8812au/5.13.6-23, 6.1.0-21-amd64, x86_64: installed`

$ iw dev:
```
phy#1
	Interface wlx00c0cab484fc
		ifindex 4
		wdev 0x100000001
		addr 00:c0:ca:b4:84:fc
		type managed
		txpower -100.00 dBm

```

Log saved after running the command `wifite --kill`:
[rtw.log](https://github.com/user-attachments/files/15587397/rtw.log)

wifite log:

```
    Interface   PHY   Driver              Chipset                                                                                                                                             
-----------------------------------------------------------------------
 1. wlx00c0cab484fcphy1  8812au              Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

 [+] Enabling monitor mode on wlx00c0cab484fc... failed

 [!] Error: Cannot find any interfaces in monitor mode

```

morrownr, 2024-06-05:
Hi @mentalpabulum 

From the README:

```
Note: Monitor mode is not supported. Linux users that want good support for monitor mode in a dual band or tri-band adapter should seek out USB WiFi adapters based on the mt7610u, mt7612u or mt7921au chipsets.
```

The monitor mode support that Realtek included in the driver is still there. It is not the greatest and I need to use my time to help work on a standards compliant driver that can go into the kernel so you are welcome to turn on monitor mode support but I will not be supporting it:

Use an edition such as `nano` to open `Makefile`. Scroll down until you find the following line:

CONFIG_WIFI_MONITOR = n

Change it to read:

CONFIG_WIFI_MONITOR = y

Then save and close the file.

Now run:

$ sudo sh install-driver.sh

Monitor mode should be active, however, my advice from the README should be considered and ...

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. Looking over the first 3 menu items could be beneficial.

mentalpabulum, 2024-06-05:
Hi @morrownr, thanks for your work! It may not be the best, but it's the only one that works (at least for me). Sorry, forgot to post, late noticed that in the new updates you turned off the functionality, I solved it a bit weird way:

`git checkout cc7a77e`

As for avoiding realtek, I agree. The current one gives me a lot of headaches, and I'm thinking of buying AWUS036AXML (MediaTek MT7921AUN). I'm only interested in monitor, injection. But from what I can [see](https://github.com/morrownr/USB-WiFi/issues/73), he's not doing well either :)

morrownr, 2024-06-05:
> But from what I can see, he's not doing well either :)

Active monitor mode is not required for injection. In fact, it is not required for anything except to speed up some things as far as I know. Regular monitor mode should server you well, at least for now and hopefully the bug in active monitor will be fixed. Which adapter you get can depend on your needs. I have an AXML and I think it is a good adapter. On the other, unless 6 GHZ is required, I usually recommend the Alfa ACHM as its range is second to none in dual band adapters and active monitor mode as well as everything else in monitor mode works well. I'd say the ACHM is pretty much the gold standard for monitor mode work and it costs less. That may change in future years but for now, that is my opinion.

I am working on a project to bring a standards compilant 8812au driver to the kernel but I can't say when that will happen.

morrownr, 2024-06-07:
Understand all. I've lived in several countries so I understand some of the challenges. I could help you go through what is available to you. Do you have access to any online stores or just local?

morrownr, 2024-06-07:
> The only problem is figuring out exactly what to buy...

I understand. This exact question is why I started this site.

I have an AXML but I also have another adapter with the same chip. Let me point you to the Main Menu:

https://github.com/morrownr/USB-WiFi

Reading menu item 1 for background may be a good idea and menu item 2 is where I keep the recommended adapters-- Plug and Play List.

I also have the EDUP EP-AX1672, which is shown in the Plug and Play List. I consider it to be a very solid lower cost tri-band adapter. It does not come with a cable like the AXML so you might want to look at getting a USB3 cable/stand to go with it so you can easily position the adapter. It does not have bluetooth support but then my AXML does not have bluetooth support either. It will show the same Active monitor mode bug as the AXML as it uses the same driver but that bug should be fixed at some point. The drivers for the Alfa ACM and ACHM do not have Active monitor mode bug so if that is important, you should consider those adapters even though they are only WiFi 5 adapters.

The bug where it appears bluetooth is involved with the AXML is hard to reproduce. It seemed to start around Feb. of this year but we have seen an increase of problems with wifi and bluetooth since the beginning of the year as so many capabilities have been added to the stack since the last LTS. Things will stabilize as time passes.

You might want to take a trip over to ZerBea's github site:

https://github.com/ZerBea/hcxdumptool

His site specializes in this things you appears to want to do. The following isse at his site goes into specific adapters:

https://github.com/ZerBea/hcxdumptool/discussions/361

Let me know if you have additional questions and please let me know what you decide to do.

@morrownr

morrownr, 2024-06-07:
You can also keep an eye open for new adapters with the new mt7925 chip. It is WiFi 7. The driver went in kernel 6.7. We are waiting impatiently for adapters. M.2 cards are starting to show. There is also the mt7927 chip that goes up to 320 MHz channel width but support for it is not in the kernel yet.

The rtw88 8812au driver may be ready for testing soon. I will post a Project issue here in this repo when it is ready for testing. We will need monitor mode testers. Do you have any other adapters? Maybe a 8821/11au based adapter?

morrownr, 2024-06-07:
> As soon as I want to buy something, I start googling and find information about poor support or bugs...

Be careful about reading too much. I've been helping usb wifi adapter users for a lot of years. It is common for bug reports to be user error. These adapters can do so many things that can be hard to set up without experience so bugs are reported against an adapter or driver when it is really user error and not a bug. People have to learn but they don't come back and explain that it was their error.

There is also a lot of different hardware out there and it takes more than a wifi driver to make these usb wifi adapters work. My point here being, many bugs that get reported against usb wifi adapters are really something else causing the problem. An example of this is the RasPi4B. RasPi made a bad decision with the chip that powers their USB3 capability. Can I make it work? Yes, and very fast. I use it mostly for AP mode. Are there things that users need to know about to make it work? Yes. Is the problem the adapters fault? No.

In my opinion, the mt7921u driver is very stable. The mt7610u and mt7612u drivers are ultra stable. They are the gold standard for monitor mode work.

> There are no guarantees that all this will be well enough supported and free of bugs in the required functionality.

How well this will likely turn out in the long run is how well the company behind the product supports its product. The product in this case are the usb wifi chips. There is a massive difference is Mediatek's usb support for its wifi chips and Realteks. Mediatek wins hands down. If all you have used is adapters with Realtek chips, I understand your reluctance.

Let me invite you to go to the Main Menu, link is above. Post a message in Issues there explaining what you want to do and ask for recommended adapters. Over time you should get plenty of replies. Your title should be something like:

`What USB WiFi adapter do you recommend for monitor mode work?`

@morrownr

morrownr, 2024-06-08:
Sure, I'll go ahead and close this issue. I would like a report on your experience with the AXML.What I hear from users greatly influences the information on this site.

closed 2024-06-08 by morrownr

morrownr, 2024-06-08:
Oh, while you are waiting, if you want to do a test on the new driver for 8812au, let me give you the location of the code:

https://github.com/lwfinger/rtw88

That is where we do the development and then tested drivers are upstreamed to the kernel. Right you won't see any documentation about the 8812au but the code is in there. I am testing managed mode this morning. The 8812au driver is not ready for public testing yet so I have not established a place to report bugs so come back here to report for now.

Installation is a manual process but I'm sure you can do it. If you have questions, come back here and ask. I know that monitor mode for the 8821au driver that we are also adding is in good shape as I have tested it but I'd be curious about a report from you regarding monitor mode on 8812au.

Regards,

@morrownr

kikitte, 2024-06-16:
> Oh, while you are waiting, if you want to do a test on the new driver for 8812au, let me give you the location of the code:
>
> https://github.com/lwfinger/rtw88
>
> That is where we do the development [...]

Thank you for your work, I've tested the rtw88 driver on master branch, it shows me an error, below image is the dmesg log, I am new to linux and it may not enough for you, if you need any other information, just tell me. Thanks again. 

[image]

morrownr, 2024-06-16:
@kikitte 

Thanks for the report. The 8812au driver in rtw88 is actually not ready for testing yet. I'll post an Issue here in this repo when it is ready for testing. Hopefully the new driver works well for all. Right now we are trying to finish up the 8821au driver. It is in really good shape so things seem to be rolling right along.