morrownr/8812au-20210820 #27: Project: Add 8812au rtw88 in-kernel driver. Need testers...

morrownr, 2024-06-23:
Greetings to anyone that reads this message.

This Issue is where we coordinate and take bug report for the new 8812au in-kernel driver.

An effort is underway to add support for the rtl8812au chipset to the rtw88 in-kernel driver series. The driver is available for testing at the following repo:

https://github.com/lwfinger/rtw88

Remember to first remove the out-of-kernel driver in this repo or whatever repo you may have installed. You can run the following to remove it if using this repo:

$ sudo sh remove-driver.sh

It is important to follow the instructions in the README at the repo with the new test driver. You may not be familiar with rtw88 in the kernel but even if you are, there are some necessary mods that you need to know about. The rtw88 that has this new driver is more advanced than the rtw88 in stable kernels as it follows wireless-next and is used to work on and develop new drivers.

We welcome you to test and report on this new driver. Your testing will help us get this new driver in the Linux kernel sooner and in better shape. If you are aware of anyone who is familiar with mac80211 drivers, please invite them as more eyes on the code is a good thing. Your ideas are most welcome. We can do this.

@morrownr

Status report:

- AP mode or power saving are not yet supported.
- USB 3 support. Please test with USB 3 ports and USB 2 ports. Even USB 1.x if you have them.

Reports go in this issue.

tcoard, 2024-06-23:
Updating to kernel 6.9.3 broke my wifi (I was on aircrack-ng/rtl8812au before), so I tried the repo you linked and it works!

System: Pop_OS!
Device: Linksys WUSB6300
USB Ports tried: 3.2 gen2, 3.2 gen1, 2.0
Notes: All ports work, I had to enter the credentials for my wifi network again when trying the 2.0 port.

Let me know if there is any more info you would like. (I don't know much about drivers, so I don't know what information is pertinent to send).

morrownr, 2024-06-23:
@dubhater

Adding you to this message to make sure you are on this new issue for the 8812au. Note that this Issue is in the 8812au repo. See first report above.

@tcoard 

> I tried the repo you linked and it works!

That is great.

> All ports work, I had to enter the credentials for my wifi network again when trying the 2.0 port.

Entering credentials again when you switch ports seems to be normal operations. Now to make sure that the driver is switching to be in the correct mode, can I get you to run the following command while in a usb2 port and then again in a usb3 port:

$ lsusb -t

Please post both sets of results. On the line for your driver, you should see 5000 if in usb3 mode and 480 if in usb2 mode. We need to make sure that is happening properly.

> Let me know if there is any more info you would like. (I don't know much about drivers, so I don't know what information is pertinent to send).

Well figure out what info we need as we go. You just keep using it and if something does not work like you think it should, let us know what is wrong and we will tell you what info we need.

Your help is greatly appreciated. It will be cool when this driver is in mainline. Our 8812au adapters will finally have a proper in-kernel driver and will be plug and play.

tcoard, 2024-06-24:
@morrownr 

Yup, I see 5000 if in usb3 mode and 480 if in usb2 mode.

usb3:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
    |__ Port 4: Dev 6, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 5000M
    |__ Port 6: Dev 2, If 0, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 6: Dev 2, If 1, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 6: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 6: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 6: Dev 2, If 4, Class=Vendor Specific Class, Driver=, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 3: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 12, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 1: Dev 9, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 1: Dev 9, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 9, If 1, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 11: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 10, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 12: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

usb2:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
    |__ Port 6: Dev 2, If 0, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 6: Dev 2, If 1, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 6: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 6: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 6: Dev 2, If 4, Class=Vendor Specific Class, Driver=, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 3: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 12, If 2, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 3: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 1: Dev 9, If 0, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 1: Dev 9, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 1: Dev 9, If 1, Class=Audio, Driver=snd-usb-audio, 12M
    |__ Port 10: Dev 20, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 480M
    |__ Port 11: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 10, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 12: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

Jake-Grafton, 2024-06-24:
The rtw8812au driver works flawlessly with my (previously useless) ALFA AWUS036ACH!

Monitor mode works! (see screenshot)
[image]

The driver also works with USB 3 and USB 2.
**USB 3 Port:**
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 5000M
```

**USB 2 Port:**
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 7, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 480M
```

kikitte, 2024-06-24:
Finally we get the kernel driver!

Model: ALFA AWUS036ACH

- **USB3 & USB2 works good.**
```
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 001: Dev 002, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 5000M
```
```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 006: Dev 002, If 0, Class=Vendor Specific Class, Driver=[none], 12M
    |__ Port 006: Dev 002, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 007: Dev 003, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 002: Dev 004, If 0, Class=Wireless, Driver=btusb, 12M
        |__ Port 002: Dev 004, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 009: Dev 006, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 480M
```
- **Some error or warning message about rtw_8812au in dmesg**
May be it would help.

[image]

dubhater, 2024-06-24:
@kikitte Someone sent a fix for those warnings: https://lore.kernel.org/linux-wireless/20240528110246.477321-1-marcin.slusarz@gmail.com/

@Jake-Grafton What is the application in your screenshot?

Redditor247, 2024-06-24:
Currently working with this driver. Installed correctly, but can't get this to go into monitor mode for whatever reason. Card being used is an Alfa AWUS036ACH, realtek 8812au driver.

```bash
sudo ip link set wlan1 down
sudo iw wlan1 set monitor none

command failed: Operation not supported (-95)
```

This will connect to Wi-Fi and hold a connection fine, but monitor mode will error out.

kikitte, 2024-06-24:
> @kikitte Someone sent a fix for those warnings: https://lore.kernel.org/linux-wireless/20240528110246.477321-1-marcin.slusarz@gmail.com/
> 
> @Jake-Grafton What is the application in your screenshot?

About monitor mode, may be something went wrong in my system, it is not related to this driver, Intel AX200 also not working with hcxdumptool but I can conform it works a few days ago.

Jake-Grafton, 2024-06-24:
> @Jake-Grafton What is the application in your screenshot?

hcxdumptool, as it is vastly superior to the aircrack-ng suite.

morrownr, 2024-06-24:
@dubhater 

The author of hcxdumptool is @ZerBea . He was helping with the testing of 8821/11au. He can likely answer questions about hcxdumptool if you have any. We may need to see if anyone has an extra 8812au based adapter that we can get into his hands. Unfortunately I only have one 8812au based adapter or I would get one in the mail to him today.

morrownr, 2024-06-24:
@Redditor247 

> tried a clean install of the experimental driver and nothing.

Let me work with you directly to get things going. Here is my email address: morrownr@gmail.com

I will delete this message after it is send as I would prefer my email address to not be too widely known.

morrownr, 2024-06-24:
@gordboy

Do you still have that 8812au based adapter? If you do, we are going to try to talk you into one of two things:

1. Help us test this new in-kernel driver or
2. Send iit to @ZerBea

ZerBea, 2024-06-24:
@morrownr 
No need to send a device, according to this screen shots, hcxdumptool is working as expected:
https://github.com/ZerBea/hcxdumptool/issues/458#issuecomment-2185787738
BTW:
Don't worry about the 160 packets dropped by kernel.
Running monitor mode and interacting with the targets (APs andCLIENTs) places very high demands on the hardware. As a result, some packets are dropped by the kernel.

I'll order this adapter:
https://www.reichelt.de/de/en/usb-3-0-dual-band-wlan-mini-stick-867-mbps-delock-12502-p208115.html?GROUPID=5839&START=0&OFFSET=16&SID=929bb67fe740d35447714c48cdad3a0042274c665a29e76c30d98&LANGUAGE=EN&&r=1

According to Delock, it should be a RTL8812AU based one:
https://www.delock.de/produkt/12502/merkmale.html?setLanguage=en

castillofrancodamian, 2024-06-24:
@morrownr The `sudo sh remove-driver.sh` command to uninstall the driver is not working. I restart the PC and still see networks in network manager.

morrownr, 2024-06-24:
@castillofrancodamian 

> The sudo sh remove-driver.sh command to uninstall the driver is not working.

If there is a previously installed 8812au driver that is not the one in this repo, not to be confused with the test driver that is in another location, then there could be problems. Run and post the results of:

$ dkms status
$ lsmod | grep 88

castillofrancodamian, 2024-06-24:
@morrownr 

```
[franco@pc ~]$ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl8812au-5.13.6/23/source/dkms.conf does not exist.
[franco@pc ~]$ lsmod | grep 88
libchacha              12288  1 chacha_x86_64
xt_nat                 12288  44
xt_conntrack           12288  2
iptable_nat            12288  9
nf_defrag_ipv4         12288  1 nf_conntrack
libcrc32c              12288  2 nf_conntrack,nf_nat
xt_addrtype            12288  2
iptable_filter         12288  2
stp                    12288  1 bridge
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
cmac                   12288  2
algif_hash             12288  1
algif_skcipher         12288  1
intel_tcc_cooling      12288  0
crct10dif_pclmul       12288  1
crc32_pclmul           12288  0
polyval_clmulni        12288  0
polyval_generic        12288  1 polyval_clmulni
ac97_bus               12288  1 snd_soc_core
8812au               3256320  0
btmtk                  12288  1 btusb
hid_generic            12288  0
iTCO_vendor_support    12288  1 iTCO_wdt
cfg80211             1363968  1 8812au
mdio_devres            12288  1 r8169
pcspkr                 12288  0
intel_pmc_core        122880  0
mac_hid                12288  0
pkcs8_key_parser       12288  0
ext4                 1146880  2
crc32c_generic         12288  0
crc16                  12288  2 bluetooth,ext4
vivaldi_fmap           12288  1 atkbd
```

morrownr, 2024-06-24:
Try:

$ sudo modprobe -r 8812au

Then reboot and run:

$ lsmod | grep 8812au

Post the result. Maybe that will uninstall it.

castillofrancodamian, 2024-06-24:
@morrownr I'm still seeing available networks.

```
[franco@pc ~]$ lsmod | grep 8812au
8812au               3256320  0
cfg80211             1363968  1 8812au
```

```
[franco@pc ~]$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 003: Dev 002, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
    |__ Port 007: Dev 003, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 007: Dev 003, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 012: Dev 004, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 012: Dev 004, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 5000M
```

Looking at the result of `lsusb -t` it seems to be connected to USB 2, but it is not true. The adapter is connected to a USB 3 port.

morrownr, 2024-06-24:
@castillofrancodamian 

>  I searched and tried make uninstall...

sudo make uninstall should uninstall the rtw88 driver that we are testing. The `8812au` driver is not the rtw88 driver. The rtw88 driver that we are testing should show with a name of rtw_8812au.

My best guess is that dkms still has the driver installed. I am trying to think of way to solve the problem with dkms but I am not there yet. Run and post:

$ modinfo 8812au

Maybe that will give us a hint.

castillofrancodamian, 2024-06-24:
@morrownr I know they are two "different" controllers. When I installed and uninstalled I did so in its corresponding directory. It was like this: I had the `8812au` driver installed (with `sudo ./install-driver.sh`), I deleted it (with `sudo ./remove-driver.sh`), I installed `rtw_8812au` (with [this](https://github.com/lwfinger/rtw88?tab=readme-ov-file#basic-installation-for-all-distros-) guide), i wanted to uninstall it, but I don't remember if `make uninstall` actually uninstalled it (apparently it did). I waited for the kernel to be updated and deleted. I reinstalled `8812au` and now it is not possible to remove it.

```
[franco@pc ~]$ modinfo 8812au
filename:       /lib/modules/6.9.6-arch1-1/updates/dkms/8812au.ko.zst
import_ns:      VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver
version:        v5.13.6-23-g232107d9b.20210820
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     583BF86F5DB8C7CC3A86DCB
alias:          usb:v7392pA822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2604p0012d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0103d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0101d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p805Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB30d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3316d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3315d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3313d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1740p0100d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp9097d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p003Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1058p0632d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0022d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0074d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p17D2d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9051d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0789p016Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0586p3426d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0952d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p025Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0409p0408d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211
retpoline:      Y
name:           8812au
vermagic:       6.9.6-arch1-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        14:B8:CF:91:D6:06:73:41:60:2C:0F:62:3C:47:DF:40:77:54:49:93
sig_hashalgo:   sha512
signature:      37:78:C0:0E:15:8B:4A:FA:96:4E:30:38:18:B5:FB:38:5A:18:1A:47:
                D5:9E:1E:18:FF:B9:84:A5:31:E2:E5:DE:17:EA:A6:0E:9F:16:55:E7:
                9C:B5:FE:26:11:49:74:78:ED:14:8B:EF:47:2D:C3:EA:6F:71:61:36:
                3E:CA:A0:FB:1B:E7:74:A4:2C:C6:8B:E8:6D:E2:8F:7C:18:52:40:4E:
                A6:7F:7A:E6:5A:3B:EB:C6:2C:D0:84:97:48:12:87:B4:5B:0A:8C:60:
                50:28:0E:9E:68:68:D2:02:69:81:D8:B0:3F:3F:70:1C:AC:BB:24:23:
                5D:61:13:A0:27:AE:7D:4F:03:BD:91:9D:07:56:0A:E4:F5:FA:26:D5:
                AB:A9:B3:E7:87:5A:D8:44:AB:5D:18:C1:A5:27:BA:01:B0:61:D2:0A:
                C3:CF:AA:F5:F7:9A:AD:19:74:B0:1D:E4:AB:55:52:13:E8:F5:88:D7:
                FB:81:A6:BD:6D:14:0F:0C:59:25:9E:B3:88:00:96:4C:BE:3E:45:A3:
                3F:F8:7E:D9:92:3C:13:67:6B:A7:1C:F2:C5:9E:F1:64:98:73:9C:72:
                A2:B4:8E:5E:BF:56:34:5C:AB:DB:2F:F6:3B:08:48:B4:22:5D:1E:2D:
                66:CD:23:CD:3E:A4:38:6D:C8:90:AB:D1:A6:49:5C:21
parm:           rtw_wireless_mode:int
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_lps_level:The default LPS level (int)
parm:           rtw_lps_chk_by_tp:int
[... 93 lines cut ...]
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_b:5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
[... cut ...]
parm:           rtw_recvbuf_nr:Preallocated number of struct recv_buf (int)
parm:           rtw_phydm_ability:uint
parm:           rtw_halrf_ability:uint
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int
parm:           rtw_en_tdls:int
parm:           rtw_en_dyn_rrsr:int
parm:           rtw_rrsr_value:int
parm:           rtw_scan_interval_thr:Threshold used to judge if scan request comes from scan UI, unit is ms. (uint)
```

dubhater, 2024-06-24:
As a workaround, you can blacklist 8812au.

castillofrancodamian, 2024-06-24:
@dubhater What would that be and how should I do it?

dubhater, 2024-06-24:
Like that:

```sh
echo "blacklist 8812au" > /etc/modprobe.d/no-8812au.conf 
```

8812au should not get loaded anymore unless you load it yourself.

castillofrancodamian, 2024-06-24:
@dubhater But the problem is that the driver is still installed, and if I want to try the new rtw_8812au again, maybe there will be more problems.

morrownr, 2024-06-25:
@castillofrancodamian 

Let me suggest that you contact me via email: morrownr@gmail.com

We can sort this out. What @dubhater is suggesting should work but I understand you want a cleaner solution. We can get it done and then we can come back here.

castillofrancodamian, 2024-06-25:
@alkisg I no longer see networks in network manager.

```
[franco@pc ~]$ modinfo 8812au
modinfo: ERROR: Module 8812au not found.
[franco@pc ~]$ lsmod | grep 8812au
[franco@pc ~]$ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/rtl8812au-5.13.6/23/source/dkms.conf does not exist.
[franco@pc ~]$ lsmod | grep 88
libchacha              12288  1 chacha_x86_64
xt_nat                 12288  44
xt_conntrack           12288  2
iptable_nat            12288  9
nf_defrag_ipv4         12288  1 nf_conntrack
libcrc32c              12288  2 nf_conntrack,nf_nat
xt_addrtype            12288  2
iptable_filter         12288  2
stp                    12288  1 bridge
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
cmac                   12288  2
algif_hash             12288  1
algif_skcipher         12288  1
intel_tcc_cooling      12288  0
ac97_bus               12288  1 snd_soc_core
crct10dif_pclmul       12288  1
crc32_pclmul           12288  0
polyval_clmulni        12288  0
polyval_generic        12288  1 polyval_clmulni
btmtk                  12288  1 btusb
hid_generic            12288  0
iTCO_vendor_support    12288  1 iTCO_wdt
mdio_devres            12288  1 r8169
pcspkr                 12288  0
intel_pmc_core        122880  0
mac_hid                12288  0
pkcs8_key_parser       12288  0
ext4                 1146880  2
crc32c_generic         12288  0
crc16                  12288  2 bluetooth,ext4
vivaldi_fmap           12288  1 atkbd
[franco@pc ~]$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 003: Dev 002, If 0, Class=Vendor Specific Class, Driver=[none], 480M
    |__ Port 007: Dev 003, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 007: Dev 003, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 012: Dev 004, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 012: Dev 004, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 5000M
```

sisi399, 2024-06-25:
I've noticed there's ongoing development of a mac80211 driver for the RTL8812AU chip. As someone who's been following this topic casually, I'm hoping for some insights:

1. Can someone explain the recent developments in layman's terms? What significant changes have occurred?

2. Historical context: In the past, many issues with RTL8812AU-based devices were attributed to Realtek not releasing mac80211-compatible source code. Has this situation changed?

3. What prompted this development? Did Realtek change their policy and release source code, or is this an independent effort?

4. I've seen mentions of a developer named "lwfinger" in relation to this project. Are they affiliated with Realtek, or are they an independent contributor?

5. Was developing a mac80211 driver always technically feasible, and the delay was due to a lack of developer expertise or interest? Or were there genuine technical or legal barriers that have only recently been overcome?

I've attempted to research these questions independently but haven't found comprehensive answers. As a novice in this field, I'd greatly appreciate any insights from those more knowledgeable about the situation.

Thank you for any information or clarification you can provide!

morrownr, 2024-06-25:
Hi @sisi399 

> I've noticed there's ongoing development of a mac80211 driver for the RTL8812AU chip.

You are at the right place. Read message one in this thread for instructions on how to test.

> I've seen mentions of a developer named "lwfinger" in relation to this project. Are they affiliated with Realtek, or are they an independent contributor?

Larry Finger passed away last Friday. RIP Larry. He will be greatly missed but we are trying to carry on without him. We are still using his rtw88 repo to do the work. @dubhater and I have the appreciate rights to use the site.

I am really busy right now so maybe we can get to more of your questions later. Right now, we need testers so please read the first message in thread and consider helping us by testing.

Regards,

@morrownr

morrownr, 2024-06-25:
@castillofrancodamian 

It appears that you have cleanly removed the pesky driver. If it is okay with you, I am going to delete a few messages related to the pesky driver in order to clean things up a bit. Is that okay?

castillofrancodamian, 2024-06-25:
> @castillofrancodamian
> 
> It appears that you have cleanly removed the pesky driver. If it is okay with you, I am going to delete a few messages related to the pesky driver in order to clean things up a bit. Is that okay?

There is no problem 👍🏼

dubhater, 2024-06-27:
AP mode should work now.

I also enabled power saving. @castillofrancodamian You will be disappointed:
```
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=125 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=147 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=170 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=295 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=113 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=32.2 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=158 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=76.6 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=201 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=18.3 ms
```

As far as I can tell, it's because of the way rtw88 decides when to enter power saving and for how long. The official driver makes those decisions differently. Something to ask Ping-Ke about...

ZerBea, 2024-06-27:
@dubhater 
Does this mean that a nl80211 message "NL80211_PS_DISABLED" addressed to "NL80211_ATTR_PS_STATE" will be ignored in monitor mode too?

dubhater, 2024-06-27:
Is that the same as running ``iw wlp3s0f3u2 set power_save off`` ? I don't think rtw88 is ignoring it.

ZerBea, 2024-06-27:
Thanks for that information.

Yes, it is the same - monitored by nlmon & Wireshark:
```
$ sudo iw dev wlp48s0f4u2u4 set power_save off

Frame 232: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface nlmon0, id 0
Linux netlink (cooked header)
Linux Generic Netlink protocol
Linux 802.11 Netlink
    Attribute: NL80211_ATTR_IFINDEX
        Len: 8
        Type: 0x0003, NL80211_ATTR_IFINDEX (3)
        Attribute Value: 0x00000003 (3)
    Attribute: NL80211_ATTR_PS_STATE
        Len: 8
        Type: 0x005d, NL80211_ATTR_PS_STATE (93)
        Attribute Value: NL80211_PS_DISABLED (0)

Frame 233: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface nlmon0, id 0
Linux netlink (cooked header)
Netlink message
    Netlink message header (type: Error)
    Error code: Success (0)
    Netlink message header (type: 0x0024)
```

nlmon in combination with Wireshark is a powerful instrument to debug NETLINK.
Unfortunately we can't trust it 100%. Some drivers report success but ignore the command.

castillofrancodamian, 2024-06-27:
> AP mode should work now.
>
> I also enabled power saving. @castillofrancodamian You will be disappointed:
>
> ```
> PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
> 64 bytes from 192.168.1.1: icmp_seq=1 [...]

But from what I understand, that behavior is correct (using an Android phone as reference). What I reported in the past is not the same as the ping you see. In my case there were interruptions in the use of the internet via Wi-Fi, not now.
Maybe what was happening to me was a power saving mode that was not working correctly. In addition, it had other problems, such as complete disconnections from a Wi-Fi network. I no longer notice any of the problems I noticed with the previous controller.

dubhater, 2024-06-27:
@castillofrancodamian 
>  In my case there were interruptions in the use of the internet via Wi-Fi, not now.

Oh, I didn't remember that.

> In addition, it had other problems, such as complete disconnections from a Wi-Fi network. I no longer notice any of the problems I noticed with the previous controller.

More bad news: I have interruptions in the use of the internet. For example, sometimes NetworkManager can't reach ping.archlinux.org for many seconds. If I move the RTL8812AU into a particular position, it stops receiving beacons and gets disconnected from the AP. The distance is about 3 meters, with no obstacles. The AP is on channel 48. The 2.4 GHz band seems better.

I didn't use the official driver long enough to see if it has the first problem, but I did reproduce the second problem with it. It has some code which tries to fix it, doesn't succeed entirely.

aikawayataro, 2024-06-27:
Is this the right place for a status report 😅? 
Im running rtw88 with my AWUS036ACH right now and it works pretty well!
I've made an Arch Linux package for this driver with firmware included, you can find it [here](https://aur.archlinux.org/pkgbase/rtw88-fixed-dkms-git) (it's "fixed" cuz there's already an rtw88 package, but it's unmaintained).
I will be running it for some time and will report any problems I encounter, thank you!

morrownr, 2024-06-28:
Hi @aikawayataro 

> Is this the right place for a status report 😅?

Yes

> Im running rtw88 with my AWUS036ACH right now and it works pretty well!

Nice adapter. I have one.

> I've made an Arch Linux package for this driver with firmware included...

I'm trying to think about whether putting rtw88 (the one we are using, not mainline) up on AUR is going to have the effect that you are intending. Are AUR users going to know to report problems here? If AUR users post issues at lwfinger/rtw88, I'm not sure that anyone is currently monitoring issues there. The owner, Larry Finger, passed away last week. Dubhater and I do have the rights to change the code at the site but you really don't want us handling the administrator chores as we are shorthanded anyway. It was 3 person team and now it is 2. What might work better is for you to make a post in the appropriate Arch forum that has a link to this issue and instructions to read message 1. Also, make it clear this is about testing rtl8812au based adapters. Things can get really complicated quickly and we really need to try to avoid that if we can.

Looking forward to your 8812au report.

@morrownr

aikawayataro, 2024-06-28:
> Are AUR users going to know to report problems here? If AUR users post issues at lwfinger/rtw88, I'm not sure that anyone is currently monitoring issues there.

Of course, I will leave a comment about 8812au testing with proper place for reports, sorry.

> Things can get really complicated quickly and we really need to try to avoid that if we can.

I believe there won't be this kind of problem, I only posted this package link here in my comment and I doubt it will spread much.

My deepest condolences on Larry's passing. He has truly impacted wireless Linux.

morrownr, 2024-06-28:
> My deepest condolences on Larry's passing. He has truly impacted wireless Linux.

He will be missed.

Where is your report?

castillofrancodamian, 2024-06-28:
> AP mode should work now.

I tried to try AP mode but it is not working. Is the configuration file the same as the one presented in USB-WiFi?

This file I tried:

<details><summary>/etc/hostapd/hostapd.conf</summary>
<p>

```
# /etc/hostapd/hostapd-WiFi5.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2022-09-25

# SSID
ssid=FDC
# PASSPHRASE
wpa_passphrase=1111111111
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# Country code
country_code=AR
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
ieee80211h=1

# Bridge interface
###bridge=br0
# WiFi interface
interface=wlan0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

# Various settings
#beacon_int=100
#dtim_period=2
#max_num_sta=32
macaddr_acl=0
#rts_threshold=2347
#fragm_threshold=2346
#send_probe_response=1

# Security
# auth_algs=3 is required for WPA3-SAE and WPA3-SAE Transition mode
auth_algs=1
ignore_broadcast_ssid=0
# wpa=2 is required for WPA2 and WPA3 (read the docs)
wpa=2
rsn_pairwise=CCMP
# only one wpa_key_mgmt= line should be active.
# wpa_key_mgmt=WPA-PSK is required for WPA2-AES
wpa_key_mgmt=WPA-PSK
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-SAE Transition mode
#wpa_key_mgmt=SAE WPA-PSK
# wpa_key_mgmt=SAE is required for WPA3-SAE
#wpa_key_mgmt=SAE
#wpa_group_rekey=1800
# ieee80211w=1 is required for WPA3-SAE Transition mode
# ieee80211w=2 is required for WPA3-SAE
#ieee80211w=1
# if parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
# sae_require_mfp=1 is required for WPA3-SAE Transition mode
#sae_require_mfp=1
# if parameter is not 9 set, 5 is the default value.
#sae_anti_clogging_threshold=10

# Note: Capabilities can vary even between adapters with the same chipset.
#
# Note: Only one ht_capab= line and one vht_capab= should be active. The
# content of these lines is determined by the capabilities of your adapter.
#
# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# generic setting
###ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# mediatek chipsets
# mt7612u - mt7610u
#ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
# mt7921au
##ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
#
# realtek chipsets
# rtl8812au - rtl8811au - rtl8811cu
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8812bu
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8814au
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
#

# IEEE 802.11ac
ieee80211ac=1
#
# generic setting
###vht_capab=[SHORT-GI-80]
#
# mt7610u
#vht_capab=[SHORT-GI-80][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
# mt7612u
#vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
# mt7921au (VHT Cap. 339071b2)
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
#
# rtl8812au - rtl8812bu
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8814au
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811au
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811cu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# Note: [TX-STBC-2BY1] may cause problems with some Realtek drivers

# end of hostapd-WiFi5.conf
```

</p>
</details>

aikawayataro, 2024-06-28:
> Where is your report?

Originally I just meant a status report that it's working fine.

But now, after using it for a while, I encountered a problem. After some time the wifi connection becomes unstable but still works, I just get a connection timeouts and it goes back to "normal" after a few seconds. The "broken" phase will return after some time, interval I've noticed is close to 10 minutes (sometimes it's longer).
Just unplugging wifi solves the problem for me. Interesting, but it stops working almost 4800 seconds after I plug it. (I observed this delay twice). During unstable wifi, my kernel log is spammed with these messages (`[12122.732596]` connected the device, `[16932.148366]` wifi issue begins):

```
[12122.732596] usb 4-4: reset SuperSpeed USB device number 5 using xhci_hcd
[12130.211073] rtw_8812au 4-4:1.0: Firmware version 52.14.0, H2C version 0
[12130.254278] 00000000: 29 81 0b 86 0c 00 00 00 02 00 00 00 00 00 00 00  )...............
[12130.254285] 00000010: 15 14 14 15 15 15 1b 1b 1b 1b 1b ee ee e0 ff ff  ................
[12130.254287] 00000020: ff ff 23 20 1f 1d 21 21 22 22 24 24 28 29 27 2a  ..# ..!!""$$()'*
[12130.254289] 00000030: fe ee ff ff ee ff 00 ee ff ff 17 15 14 13 13 13  ................
[12130.254291] 00000040: 1b 1b 1a 1a 1a 0e ee e0 ff ff ff ff 24 23 22 22  ............$#""
[12130.254292] 00000050: 24 20 1e 1a 19 18 1b 1e 1f 21 ed ee ff ff ee ff  $ .......!......
[12130.254293] 00000060: 00 ee ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254295] 00000070: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254296] 00000080: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254297] 00000090: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254298] 000000a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254300] 000000b0: ff ff ff ff ff ff ff ff 7f 1c 27 00 33 88 00 88  ..........'.3...
[12130.254301] 000000c0: 00 00 00 10 00 00 00 55 00 ff 03 ff ff ff ff ff  .......U........
[12130.254303] 000000d0: da 0b 12 88 41 66 42 00 c0 ca b2 7b 98 09 03 52  ....AfB....{...R
[12130.254304] 000000e0: 65 61 6c 74 65 6b 0d 03 38 30 32 2e 31 31 6e 20  ealtek..802.11n 
[12130.254305] 000000f0: 4e 49 43 08 03 31 32 33 34 35 36 ff ff ff ff ff  NIC..123456.....
[12130.254307] 00000100: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254308] 00000110: ff ff ff ff ff ff ff 21 03 03 0a ff 00 ff ff ff  .......!........
[12130.254309] 00000120: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254311] 00000130: 81 ae 98 2d 03 93 96 11 fc 8c 00 11 9b 78 4a ce  ...-.........xJ.
[12130.254312] 00000140: e0 48 e0 70 27 00 78 ea 60 d0 e1 20 32 63 4f 3c  .H.p'.x.`.. 2cO<
[12130.254314] 00000150: 12 40 92 15 8b 4a 4f 8a 15 d1 98 00 ff ff ff ff  .@...JO.........
[12130.254315] 00000160: ff ff ff ff ff ff ff ff 00 02 e1 63 12 29 03 00  ...........c.)..
[12130.254317] 00000170: cb 3f a0 e0 c2 f0 f7 cf 01 36 08 32 05 eb 94 d7  .?.......6.2....
[12130.254318] 00000180: 14 6d c8 67 80 e0 f8 80 30 80 30 38 ff ff ff ff  .m.g....0.08....
[12130.254319] 00000190: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254321] 000001a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254322] 000001b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254323] 000001c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254324] 000001d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254326] 000001e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
[12130.254327] 000001f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
...
[16932.148366] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[16954.842707] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[16977.003793] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[16999.591275] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17021.965408] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17044.446098] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17066.633808] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17088.820528] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17111.197898] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17133.278251] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17155.888865] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17177.969160] rtw_8812au 4-4:1.0: failed to get tx report from firmware
[17200.716225] rtw_8812au 4-4:1.0: failed to get tx report from firmware
```

Previous driver I used was https://github.com/aircrack-ng/rtl8812au but sometimes required me to replug my wi-fi after startup and works flawlessly after that (It got broken after recent commit and they ignore user reports so I decided to give rtw88 a shot).

morrownr, 2024-06-28:
@castillofrancodamian 

> I tried to try AP mode but it is not working. Is the configuration file the same as the one presented in USB-WiFi?

The settings for AP mode are different for rtw_8821au than for 8821au (the vendor driver) so I expect that to be the case with this driver also. I had not tested AP mode yet for this driver since Dubhater said it was not working but now I'll take a look and will post what I find.

dubhater, 2024-06-28:
@aikawayataro "failed to get tx report from firmware" is caused by the power saving. They are about 22 seconds apart. It's strange that it's so regular.

morrownr, 2024-06-28:
@castillofrancodamian 

hostapd.conf settings for rtw_8812au:

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]

Things to keep in mind:

- The settings above are based on what the driver says it can do and not me actually testing the settings. I will test but have not had time yet,

- Interestingly enough, rtw_8821au actually shows more capability than rtw_8812au. This is not what I am used to seeing with AC600 vs. AC1200.

- If the settings do not work, report what you find. rtw88 (mainline) does not have a good track record for supporting what it says it supports but this driver is an add on and I have been following exactly what it is depending on for core support. You are the first tester of AP mode in this driver.

morrownr, 2024-06-28:
@castillofrancodamian 

FYI: The above setting are for band 2 (5 GHz). If you need band 1 settings, let me know.

morrownr, 2024-06-28:
@dubhater 

Bug report:

Debian 12 with kernel 6.6
Country code: US

I am seeing the following for rtw_8812au, which is not correct:

5720 MHz [144] (disabled)

In the same machine, the mt7922 PCIe adapter is showing the following, which is correct:

5720 MHz [144] (24.0 dBm) (radar detection)

Is the rtw_8812au driver using info from the eeprom? At one time, several years ago, the disabled reading would have been correct but that channel has been allowed for several years now in the US.

aikawayataro, 2024-06-29:
> @aikawayataro "failed to get tx report from firmware" is caused by the power saving. They are about 22 seconds apart. It's strange that it's so regular.

I was too amazed at how regular the interval is. Sometimes it's not 22 seconds, but 12 or even 15 (varies between replugs). The more interesting part is that this behavior always starts 4800 seconds after I plugged it in. I will play with the power for a bit to confirm, thank you.

morrownr, 2024-06-29:
@dubhater 

Disregard the channel available bug report above. I made a mistake.

amnuhs, 2024-06-29:
cannot enable monitor mode - alfa awus036ach

morrownr, 2024-06-29:
@amnuhs 

Can you provide the steps you used to attempt to enter monitor mode? Thanks.

castillofrancodamian, 2024-06-29:
> @castillofrancodamian
>
> hostapd.conf settings for rtw_8812au:
>
> ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935] [...]

I tried again and it is not possible to connect to the AP from an Android device. It tries to get an IP and gets disconnected. I tried connecting with a static IP and it connects but immediately disconnects in a loop.

<details><summary>cat /etc/hostapd/hostapd.conf</summary>
<p>

```
[franco@pc ~]$ cat /etc/hostapd/hostapd.conf
ssid=FDC
wpa_passphrase=1111111111
hw_mode=a
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
country_code=AR
ieee80211d=1
ieee80211h=1
###bridge=br0
interface=wlan0
driver=nl80211
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]
```

</p>
</details> 

<details><summary>cat /etc/dnsmasq.conf</summary>
<p>

```
[franco@pc ~]$ cat /etc/dnsmasq.conf 
# Interface to bind to
interface=wlan0
# Specify starting_range,end_range,lease_time
dhcp-range=192.168.70.2,192.168.70.100,12h
 
# Uncomment and modify the following lines if you don't want to forward dns from the host's /etc/resolv.conf
## disables dnsmasq reading any other files like /etc/resolv.conf for nameservers
no-resolv
## dns addresses to send to the clients
server=192.168.20.1
dhcp-option=6,192.168.20.1
```

</p>
</details> 

Interface configuration

```
sudo ip link set up dev wlan0
sudo ip addr add 192.168.90.1/24 dev wlan0
sudo iptables -t nat -A POSTROUTING -o enp1s0 -j MASQUERADE
[... 318 lines cut ...]
jun 29 16:47:03 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:04 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:04 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:13 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:14 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:14 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:14 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:14 pc hostapd[3930]: handle_probe_req: send failed
jun 29 16:47:14 pc hostapd[3930]: handle_probe_req: send failed
```

</p>
</details>

morrownr, 2024-06-29:
@castillofrancodamian 

Hmmm... I am going to go ahead and terminate the current AP mode testing on my RasPi4B and see about testing rtw_8812au.

I'll let you know what I find.

morrownr, 2024-06-29:
@castillofrancodamian 

Back to you.

Looking good here so far. Here is what I am seeing when using the rtll8812au based adapter with hostapd:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.175 port 48190 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  38.5 MBytes   323 Mbits/sec    0    892 KBytes       
[  5]   1.00-2.00   sec  40.0 MBytes   336 Mbits/sec    0    892 KBytes       
[  5]   2.00-3.00   sec  38.8 MBytes   325 Mbits/sec    0    892 KBytes       
[  5]   3.00-4.00   sec  41.2 MBytes   346 Mbits/sec    0    892 KBytes       
[  5]   4.00-5.00   sec  40.0 MBytes   336 Mbits/sec    0    936 KBytes       
[  5]   5.00-6.00   sec  40.0 MBytes   336 Mbits/sec    0    936 KBytes       
[  5]   6.00-7.00   sec  38.8 MBytes   325 Mbits/sec    0    936 KBytes       
[  5]   7.00-8.00   sec  40.0 MBytes   336 Mbits/sec    0    936 KBytes       
[  5]   8.00-9.00   sec  40.0 MBytes   336 Mbits/sec    0    984 KBytes       
[  5]   9.00-10.00  sec  38.8 MBytes   325 Mbits/sec    0    984 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   396 MBytes   332 Mbits/sec    0             sender
[  5]   0.00-10.03  sec   394 MBytes   329 Mbits/sec                  receiver

```

Here is the exact hostapd.conf that I am using:

```
# /etc/hostapd/hostapd-WiFi5.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2022-09-25

# SSID
ssid=myPI-WiFi5
# PASSPHRASE
wpa_passphrase=myPW1234
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel
channel=149
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=155
# Country code
country_code=US
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
#ieee80211h=1

# Bridge interface
bridge=br0
# WiFi interface
#interface=wlx00c0caac47c7
interface=wlx00c0caaae8ba

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

# Various settings
#beacon_int=100
#dtim_period=2
#max_num_sta=32
[... 90 lines cut ...]

# band 2: rtw88 rtl8811au rtl8811cu
# VHT Capabilities (0x03d07122):
# 	Max MPDU length: 11454
# 	Supported Channel Width: neither 160 nor 80+80
# 	short GI (80 MHz)
# 	SU Beamformee
# 	MU Beamformee
# 	+HTC-VHT
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][SU-BEAMFORMEE][MU-BEAMFORMEE][HTC-VHT]
# rtw88 rtl8812au
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]

# rtl8811cu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# Note: [TX-STBC-2BY1] may cause problems with some Realtek drivers

# end of hostapd-WiFi5.conf

```

The networking setup can be found in my AP guide on the Main Menu.

@morrownr

morrownr, 2024-06-29:
@dubhater 

rtw_8812au initial AP mode report:

The compilation is based on the rtw88 repo as it was about 10 minutes ago... in other words, I did a clean pull of the repo for AP testing.

Very nice so far. Here are the appropriate parts from the log:

```
$ sudo dmesg | grep 8812
[    2.295892] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[    8.082496] rtw_8812au 1-1.2:1.0: Firmware version 52.14.0, H2C version 0
[    8.314719] rtw_8812au 1-1.2:1.0: switching to USB 3 mode
[    8.319343] usbcore: registered new interface driver rtw_8812au
[    8.641745] usb 2-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[    8.647567] rtw_8812au 2-2:1.0: Firmware version 52.14.0, H2C version 0
[    8.806345] rtw_8812au 2-2:1.0 wlx00c0caaae8ba: renamed from wlan0
[   17.103792] rtw_8812au 2-2:1.0 wlx00c0caaae8ba: entered allmulticast mode
[   17.103990] rtw_8812au 2-2:1.0 wlx00c0caaae8ba: entered promiscuous mode

```

I brought the system up from a cold boot and checked USB mode. It is in USB3 mode.

So far, looking good.

morrownr, 2024-06-29:
@dubhater @castillofrancodamian 

I am running into problems with AP mode now. Let me get an idea of what is going on and I'll let you know.

castillofrancodamian, 2024-06-30:
@morrownr I got all these errors. There were complete disconnections on several occasions using it as a station.

```
[franco@pc ~]$ sudo dmesg | grep 8812
[sudo] contraseña para franco: 
[  419.176855] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[  704.530924] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 1512.859028] rtw_8812au 2-1:1.0: failed to validate firmware
[ 1512.859040] rtw_8812au 2-1:1.0: failed to download firmware
[ 1512.859046] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 1512.859052] rtw_8812au 2-1:1.0: leave idle state failed
[ 1512.859574] rtw_8812au 2-1:1.0: failed to leave ips state
[ 1512.859579] rtw_8812au 2-1:1.0: failed to leave idle state
[ 1514.332867] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5237.945364] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[ 5237.991465] rtw_8812au 1-1:1.0: switching to USB 3 mode
[ 5238.527990] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[ 5239.300989] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5255.615421] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5268.387101] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5676.960012] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5799.409474] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 5842.555484] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 6250.944464] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 6660.936640] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 7066.929680] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 7368.000868] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 7593.886391] rtw_8812au 1-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 7618.105414] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[ 7618.158059] rtw_8812au 1-1:1.0: switching to USB 3 mode
[ 7618.528325] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[ 7619.350444] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 7620.030500] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 8032.327751] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 8445.926404] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 8858.925341] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 9271.912750] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[ 9684.910829] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[10097.908928] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[10510.899357] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[10923.894211] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[10985.193245] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11010.912711] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11011.622584] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11016.119424] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11134.933573] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11541.878512] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[11663.107905] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[12462.029347] rtw_8812au 2-1:1.0: failed to validate firmware
[12462.029360] rtw_8812au 2-1:1.0: failed to download firmware
[12462.029366] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[12462.029372] rtw_8812au 2-1:1.0: leave idle state failed
[12462.029895] rtw_8812au 2-1:1.0: failed to leave ips state
[12462.029900] rtw_8812au 2-1:1.0: failed to leave idle state
[12463.803009] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[13756.227491] rtw_8812au 1-1:1.0: Firmware version 52.14.0, H2C version 0
[13756.279073] rtw_8812au 1-1:1.0: switching to USB 3 mode
[13756.586499] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[13757.370699] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[13758.023224] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[14164.841398] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[14570.832862] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[14980.837919] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[15390.838801] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[15771.621439] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[16177.813762] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[16582.812704] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[16802.883457] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[17287.917312] rtw_8812au 2-1:1.0: rtw8821a_power_off: bailing because RTW_FLAG_POWERON
[17288.279927] rtw_8812au 2-1:1.0: Firmware version 52.14.0, H2C version 0
[... 26 lines cut ...]
[... cut ...]
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 007: Dev 002, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 007: Dev 002, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 012: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 012: Dev 003, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 001: Dev 005, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 5000M
```

dubhater, 2024-06-30:
@castillofrancodamian have you tried different USB ports?

dubhater, 2024-06-30:
I feel like the disconnections in the 5 GHz band are less likely with this:
```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8821a.c b/drivers/net/wireless/realtek/rtw88/rtw8821a.c
index 99100ab3681c..cbf2249ed97c 100644
--- a/rtw8821a.c
+++ b/rtw8821a.c
@@ -4338,6 +3634,9 @@ static void rtw8821a_pwr_track(struct rtw_dev *rtwdev)
 {
 	struct rtw_dm_info *dm_info = &rtwdev->dm_info;
 
+	if (rtwdev->chip->id == RTW_CHIP_TYPE_8812A)
+		return;
+
 	if (!dm_info->pwr_trk_triggered) {
 		rtw_write_rf(rtwdev, RF_PATH_A, RF_T_METER,
 			     GENMASK(17, 16), 0x03);
```

castillofrancodamian, 2024-07-01:
> @castillofrancodamian have you tried different USB ports?

Yes, even those from another device.

```
[root@netbook franco]# uname -a
Linux netbook 6.9.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 28 Jun 2024 04:32:50 +0000 x86_64 GNU/Linux
```

```
[root@netbook franco]# lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ehci-pci/8p, 480M
    |__ Port 001: Dev 002, If 0, Class=Mass Storage, Driver=uas, 480M
    |__ Port 004: Dev 003, If 0, Class=Vendor Specific Class, Driver=rtw_8812au, 480M
    |__ Port 008: Dev 004, If 0, Class=Mass Storage, Driver=ums-realtek, 480M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=uhci_hcd/2p, 12M
/:  Bus 005.Port 001: Dev 001, Class=root_hub, Driver=uhci_hcd/2p, 12M
```

```
[root@netbook franco]# dmesg | grep 8812
[   30.204284] rtw_8812au 2-4:1.0: Firmware version 52.14.0, H2C version 0
[   30.399434] usbcore: registered new interface driver rtw_8812au
[   30.457039] rtw_8812au 2-4:1.0 wlp0s29f7u4: renamed from wlan0
[  499.608741] rtw_8812au 2-4:1.0: rtw8821a_power_off: RTW_FLAG_POWERON not set, powering off anyway
[  506.824460] rtw_8812au 2-4:1.0: failed to poll offset=0x5f8 mask=0xff value=0x0
[  605.188488] rtw_8812au 2-4:1.0: rtw8821a_power_off: RTW_FLAG_POWERON not set, powering off anyway
[  612.130408] rtw_8812au 2-4:1.0: failed to poll offset=0x5f8 mask=0xff value=0x0
[  716.011260] rtw_8812au 2-4:1.0: rtw8821a_power_off: RTW_FLAG_POWERON not set, powering off anyway
[  722.801878] rtw_8812au 2-4:1.0: failed to poll offset=0x5f8 mask=0xff value=0x0
```

```
ssid=FDC
wpa_passphrase=1111111111
hw_mode=a
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
country_code=AR
ieee80211d=1
ieee80211h=1
###bridge=br0
interface=wlp0s29f7u4
driver=nl80211
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]
```

```
[root@netbook franco]# iw dev
phy#1
        Interface wlp0s29f7u4
                ifindex 4
                wdev 0x100000001
                addr 18:d6:c7:11:b3:6c
                ssid FDC
[... 143 lines cut ...]
jul 01 18:24:59 netbook hostapd[716]: wlp0s29f7u4: AP-ENABLED
jul 01 18:26:30 netbook hostapd[716]: wlp0s29f7u4: interface state ENABLED->DISABLED
jul 01 18:26:30 netbook hostapd[716]: wlp0s29f7u4: AP-DISABLED
jul 01 18:26:30 netbook hostapd[716]: wlp0s29f7u4: CTRL-EVENT-TERMINATING
jul 01 18:26:30 netbook hostapd[716]: nl80211: deinit ifname=wlp0s29f7u4 disabled_11b_rates=0
jul 01 18:26:30 netbook systemd[1]: Stopping Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
jul 01 18:26:37 netbook systemd[1]: hostapd.service: Deactivated successfully.
jul 01 18:26:37 netbook systemd[1]: Stopped Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
jul 01 18:26:37 netbook systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
jul 01 18:26:38 netbook hostapd[726]: wlp0s29f7u4: interface state UNINITIALIZED->COUNTRY_UPDATE
jul 01 18:26:41 netbook hostapd[726]: wlp0s29f7u4: interface state COUNTRY_UPDATE->ENABLED
jul 01 18:26:41 netbook hostapd[726]: wlp0s29f7u4: AP-ENABLED
jul 01 18:28:21 netbook hostapd[726]: wlp0s29f7u4: interface state ENABLED->DISABLED
jul 01 18:28:21 netbook hostapd[726]: wlp0s29f7u4: AP-DISABLED
jul 01 18:28:21 netbook hostapd[726]: wlp0s29f7u4: CTRL-EVENT-TERMINATING
jul 01 18:28:21 netbook hostapd[726]: nl80211: deinit ifname=wlp0s29f7u4 disabled_11b_rates=0
jul 01 18:28:21 netbook systemd[1]: Stopping Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
jul 01 18:28:28 netbook systemd[1]: hostapd.service: Deactivated successfully.
jul 01 18:28:28 netbook systemd[1]: Stopped Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
jul 01 18:28:28 netbook systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
jul 01 18:28:30 netbook hostapd[750]: wlp0s29f7u4: interface state UNINITIALIZED->COUNTRY_UPDATE
jul 01 18:28:30 netbook hostapd[750]: wlp0s29f7u4: interface state COUNTRY_UPDATE->HT_SCAN
jul 01 18:28:34 netbook hostapd[750]: wlp0s29f7u4: interface state HT_SCAN->ENABLED
jul 01 18:28:34 netbook hostapd[750]: wlp0s29f7u4: AP-ENABLED
```

[100 more replies]