morrownr/8812au-20210820 #28: Problem about exports duplicate symbol

Lingzilla, 2024-07-09:
Hi, I recently got some 8812AU cards for free,
the hardware platform I use is Nvidia Jetson Orin Nano
the kernel version is: `5.10.104-tegra #34 SMP PREEMPT Wed May 10 15:41:45 CST 2023 aarch64 aarch64 aarch64 GNU/Linux`

dmesg output when I plug in the card:
```
[  435.520414] usb 1-3.4: USB disconnect, device number 4
[  435.712937] tegra-xusb 3610000.xhci: entering ELPG done
[  449.838755] tegra-xusb 3610000.xhci: Firmware timestamp: 2021-12-01 05:27:10 UTC, Version: 80.03 release
[  450.294496] usb 1-3.4: new high-speed USB device number 5 using tegra-xusb
[  450.398613] usb 1-3.4: New USB device found, idVendor=0bda, idProduct=881a, bcdDevice= 0.00
[  450.398621] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  450.398625] usb 1-3.4: Product: 802.11n NIC
[  450.398629] usb 1-3.4: Manufacturer: Realtek
[  450.398632] usb 1-3.4: SerialNumber: 123456
[  450.430926] cfg80211: exports duplicate symbol __cfg80211_alloc_event_net_skb (owned by kernel)
```

seems likely the module is duplicate with some other things ?

morrownr, 2024-07-09:
Hi @Lingzilla 

We have some options with your adapters. Let me ask some questions:

Do you have the option of using a more modern distro?

You also have the option of trying a new driver that is in testing that is intended to be upstreamed into the Linux kernel. You would first need to uninstall this driver by:

$ sudo sh remove-driver.sh

...and then go to the follow Issue and read the first message:

https://github.com/morrownr/8812au-20210820/issues/27

Lingzilla, 2024-07-15:
Because I am using a nvidia's proprietary hardware, so the only distro for this is Jetson Linux, and the newest version R36 of it which released two months ago is based on Ubuntu jammy (22.04)
And the kernel version is `Linux nvidia-new 5.15.136-tegra #1 SMP PREEMPT Tue Jul 9 01:22:23 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`
So sorry about that, I can not get the newer kernel to help the test.

morrownr, 2024-07-15:
> So sorry about that, I can not get the newer kernel to help the test.

Those were two issues. You can help test. Go to the following issue and read the first message:

https://github.com/morrownr/8812au-20210820/issues/27

>  the newest version R36 of it which released two months ago is based on Ubuntu jammy (22.04)

It would be best if you can backup what you need to keep and do a clean installation of this latest release. The closer to modern you get, the easier it is for me and others to help.