morrownr/8812au-20210820 #34: Driver crashes PC on Arch Linux 6.10.8-arch1-1

c0ncatenate, 2024-09-07:
Hello,

I am currently on **Arch Linux** kernel **6.10.8-arch1-1** and I have an **Alfa AWUS036ACS AC600 Wireless Adapter**. [DeviWiki](https://deviwiki.com/wiki/Iconnect_AWUS036ACS) suggests that my adapter uses the RTL8812AU driver. I've installed the driver using the `install.sh` script provided and everything installs as intended.

The device seems to recognize the driver and the LED lights up. I can see it in my WiFi devices list. However, when I try to use an app like Wireshark, it crashes my entire PC. When I try using tools such as `sudo airodump-ng wlan1`, it is unable to find any APs around me. I believe the driver is not updated to the latest kernel on Arch and it's breaking something in the background. I have also tried the driver from [Aircrack-NG](https://github.com/aircrack-ng/rtl8812au) but it doesn't work for me either. If anyone has an updated driver/kernel that I can use or a workaround to get it working, please let me know. I am devastated. :/

If you need any additional information from me, I'd be happy to provide. Thank you!

morrownr, 2024-09-07:
Hi @c0ncatenate 

I think you need a different driver. Run the following and post the results:

$ lsusb

getong, 2024-10-11:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8002 Intel Corp. 8 channel internal hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 8087:800a Intel Corp. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

uname
```
uname -a 
Linux archlinux 6.10.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 12 Sep 2024 17:21:02 +0000 x86_64 GNU/Linux
```

getong, 2024-10-11:
```
Linux archlinux 6.11.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Oct 2024 20:11:06 +0000 x86_64 GNU/Linux
```
not work

morrownr, 2024-10-11:
@c0ncatenate 

Alfa AWUS036ACS AC600 Wireless Adapter

This adapter uses the rtl8811au chip. Options for drivers:

https://github.com/morrownr/8821au-20210708
https://github.com/morrownr/8821au-20210708/issues/133

Note: The second driver has been used for dev to get a good standards compliant driver in the kernel. The driver is currently under review and should be in an upcoming kernel such as 6.13. Remember to uninstall any previously installed drivers.

morrownr, 2024-10-11:
@getong 

It is not clear to me what your problem is. Recomment that you start a new issue and tell me what the problem is.

getong, 2024-10-12:
> @getong
> 
> It is not clear to me what your problem is. Recomment that you start a new issue and tell me what the problem is.

not work in archlinux 6.11.3-arch1-1

morrownr, 2024-10-12:
@getong 

There is a pinned issues called `Problem Reporting Guidelines - Please Read.` Please follow the directions in that issue.

Without information to help determine the problem, there is little that can be done.