morrownr/8812au-20210820 #35: 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

thevirtualfer, 2024-09-15:
Hello !

Any plan to insert this device ?

Vendor ID: 2357
Prod ID: 011e

You requested the web link to the device, but I really don't know how to find this.

All I know is the output of lsusb, whith is on the title.

morrownr, 2024-09-15:
Hi @thevirtualfer 

> Any plan to insert this device ?

No, this would be the wrong driver. See:

https://github.com/morrownr/8821au-20210708

Muji1995, 2024-10-27:
can anyone help me with the drivers TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano] KALI 6.11.2-amd64

morrownr, 2024-10-27:
Hi @Muji1995 

You have options:

1. https://github.com/lwfinger/rtw88

I help at that site. In fact, the rtl8811au driver at this site is nearly finished with review and approval to go in the Linux kernel. I think it will make it into kernel 6.13. In stall the driver in the link to get a preview.

2. https://github.com/morrownr/8821au-20210708

That is the out-of-kernel driver at this site.

3. The Main Menu for this site is located:

https://github.com/morrownr/USB-WiFi?tab=readme-ov-file

The second line under menu item 2 is `Recommended adapters for Kali Linux".. You will likely save yourself a lot of frustration if you get an adapter on that list.