morrownr/8812au-20210820 #37: install-driver.sh fails when SecureBoot is enabled and DKMS is not used

thonigmann, 2024-10-03:
This is Ubuntu 24.04 running in a virtual machine, but I'm sure this will occur on real hardware as well.
Secure Boot is enabled and DKMS is **not** installed.
Calling sudo ./install-driver.sh fails with a message "make: sign: File not found"

Apparently, the problem is the definition of the target sign-install in Makefile, which orders make to call a program called sign that doesn't exist.
```
sign-install:
        sign install

```
Changing the definition to making the targets *sign* and *install* fixes the issue.
```
sign-install: sign install
```

Here is the complete output of install-driver.sh:
```
$ sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20240429
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 3960504 (total system memory)
: 6.8.0-45-generic (kernel version)
: gcc-13 (version of gcc used to compile the kernel)
: gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
: SecureBoot enabled
: ---------------------------

Checking for previously installed drivers...
Uninstalling a non-dkms installation:
/lib/modules/6.8.0-45-generic/kernel/drivers/net/wireless/8812au.ko
Deleting 8812au.conf from /etc/modprobe.d
Deleting source files from /usr/src/rtl8812au-5.13.6-23
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-45-generic/build M=/home/thomas/src/8812au-20210820-main  modules
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-45-generic“ wird betreten
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_cmd.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_security.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_debug.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_io.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_ioctl_query.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_ioctl_set.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_ieee80211.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_mlme.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_mlme_ext.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_mi.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_wlan_util.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_vht.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_pwrctrl.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_rf.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_chplan.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/monitor/rtw_radiotap.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_recv.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_sta_mgt.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_ap.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/wds/rtw_wds.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/mesh/rtw_mesh.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/thomas/src/8812au-20210820-main/core/rtw_xmit.o
[... 121 lines cut ...]
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/rtl8812a/halhwimg8812a_mac.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/rtl8812a/halhwimg8812a_bb.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/rtl8812a/halhwimg8812a_rf.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/halrf/rtl8812a/halrf_8812a_ce.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/rtl8812a/phydm_regconfig8812a.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /home/thomas/src/8812au-20210820-main/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /home/thomas/src/8812au-20210820-main/platform/platform_ops.o
  LD [M]  /home/thomas/src/8812au-20210820-main/8812au.o
  MODPOST /home/thomas/src/8812au-20210820-main/Module.symvers
  CC [M]  /home/thomas/src/8812au-20210820-main/8812au.mod.o
  LD [M]  /home/thomas/src/8812au-20210820-main/8812au.ko
  BTF [M] /home/thomas/src/8812au-20210820-main/8812au.ko
Skipping BTF generation for /home/thomas/src/8812au-20210820-main/8812au.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-45-generic“ wird verlassen
: SecureBoot enabled - read FAQ about SecureBoot
sign install
make: sign: Datei oder Verzeichnis nicht gefunden
make: *** [Makefile:2534: sign-install] Fehler 127
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

thonigmann, 2024-10-06:
Yes, the script works now as expected.

closed 2024-10-06 by thonigmann

morrownr, 2024-10-06:
Thank you very much for reporting this and testing.