morrownr/8812au-20210820 #38: Advice: AWUS036ACH, Raspberry pi 4 and kismet

cyberpunked1985, 2024-10-06:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

None

### uname

Successful

### lsusb

Successful, Device shows

### rfkill

Successful

### dkms

```text
Successful
```

### iw

```text
Successful, Phy1 shows up
```

### git status

```text
Successful
```

### install-driver.sh

```text
Successful
```

### What happened?

With further research and looking at this hub, I may have got the wrong devices for the job?
I am trying to utilize my RPI 4B as a Kismet Wifi scanner, I open to any OS/Kernel recommendation. But from my understanding is that Monitor mode is picky with the type of chip in my device. I have attempted a few different OS and getting monitor mode to enable, I have been successful getting monitor mode to enable with airmon github driver but I still find when I get Kismet up and running no Wifi devices are being discovered. Are there any Recommendations to get my current hardware functional as a kismet scanner? Should I go with an older version of Kali/Kernel or and older version of RPIOS/Kernel? Or am I SOL with my current hardware?  Any advice is greatly appreciated, I may be doing something incorrectly?

morrownr, 2024-10-07:
@cyberpunked1985 

Let me start by saying that my Monitor mode work is limited these days due to other projects but I do listen to those who are heavy Monitor mode users.

On the Main Menu:

https://github.com/morrownr/USB-WiFi

...under menu item 2, you will find a link called `Recommended Adapters for Kali Linux`. Select it. You will find good suggestions. To help with a specific suggestion, we would need to know if you want long range, small or large adapter, whether you need Active Monitor mode and any other specific needs you have.

> With further research and looking at this hub, I may have got the wrong devices for the job?

You may have picked an adapter with a chip that is a challenge for what you are trying to do. Notice in the above list about adapters for Kali, no Realtek based adapters are listed. This is because Realtek's USB WiFi support is lacking. I am aware that many online `Best adapters for Kali` lists show your adapter but those lists are simply wrong. I do my best to get good information out there but there is so much bad information that it is hard to compete.

> Should I go with an older version of Kali/Kernel or and older version of RPIOS/Kernel? 

The current 64 bit RasPiOS with kernel 6.6 should work fine.

> Any advice is greatly appreciated, I may be doing something incorrectly?

I'd say that your needs seem to require a different adapter. Like I said above, give us more info and take a look at the adapters that are recommended for Kali and we can hopefully help you.

cyberpunked1985, 2024-10-07:
Ok thank you