morrownr/8812au-20210820 #41: [Bug]: WPA2 not working - Falling back to WEP or WPA[1] Mode

uhlat, 2024-10-27:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux sihnon 6.11.5-arch1-1 #2 SMP PREEMPT_DYNAMIC

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter Bus 003 Device 003: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

0: phy2: Wireless LAN 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 2: phy1: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
rtl8812au/5.13.6-23, 6.11.5-arch1-1, x86_64: installed (original_module exists)
zfs/2.2.6, 6.11.5-arch1-1, x86_64: installed
```

### iw

```text
phy#0
	Interface wlp1s0
		ifindex 7
		wdev 0x1
		addr 
		ssid 
		type AP
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	405	0	0	0	0	42906		405
phy#2
	Interface wlp0s16u1
		ifindex 6
		wdev 0x200000001
		addr 
		ssid 
		type AP
		channel 149 (5745 MHz), width: 40 MHz, center1: 5755 MHz
		txpower 14.00 dBm
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.
```

### install-driver.sh

```text
: ---------------------------
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 3993088 (total system memory)
: 6.11.5-arch1-1 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 14.2.1 20240910
: dkms-3.1.1
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Sign command: /usr/lib/modules/6.11.5-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)............................................................................................................................................................................. done.
Signing module /var/lib/dkms/rtl8812au/5.13.6-23/build/8812au.ko
Cleaning build area... done.
The driver was built by dkms successfully.
: ---------------------------

Found pre-existing /usr/lib/modules/6.11.5-arch1-1/updates/dkms/8812au.ko.zst, archiving for uninstallation
Installing /usr/lib/modules/6.11.5-arch1-1/updates/dkms/8812au.ko.zst
Running depmod....... done.
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates to this driver SHOULD be performed before major
      upgrades such as kernel 6.5 to 6.6.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 3 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] n
```

### What happened?

WPA2 is not working in AP mode on my Alfa AWUS036ACH. Most likely just an error on my side, but I could not figure out which, so just in case:

- Using hostapd 2.11 on ArchLinux 6.11.5 (and 6.10.10) to create an ieee80211ac Access Point
- WPA2 is not working
- WEP is automatically selected.
- When I set wpa=1 or wpa=2 WEP ist selected, not WPA or WPA2
- If I set wpa=3 WPA[1] is selected

I tried different configurations, but the result is always the same. Also tried the driver maintained by the aircrack-ng guys, which has the same problem for me.

Other Adapters/Chipsets on this specific system work fine (in this regard).

[myhostapd.txt](https://github.com/user-attachments/files/17533597/myhostapd.txt)

uhlat, 2024-10-27:
It seems to be related to hostapd. Downgrading from 2.11 to 2.10 fixed that problem for me. However, this specific one was only relevant to my rtl8812au device.

So might still be something with the driver?

morrownr, 2024-10-27:
Hi @uhlat 

> WPA2 is not working in AP mode

I see.

> So might still be something with the driver?

Probably not.

Can you post your hostapd.conf file?

uhlat, 2024-10-27:
Hi there! I attached the hostapd.conf to the top post,

morrownr, 2024-10-27:
Oh. I missed it.

> wpa=3

change that line to:

wpa=2

delete the following lines unless you are sure you need them:

```
require_ht=1
require_vht=1
```

That should get you close.

uhlat, 2024-10-28:
Thank you for your help - and while at it for all your work and wonderful resources!

Your suggested changes also result in the above mentioned problem with hostapd 2.11. However with 2.10 it works as expected.

uhlat, 2024-10-28:
A workaround is using hostapd **2.10** (_not_ 2.11) .

Since this is most likely hostapd related I'll just close this issue here and see what the next hostapd version brings.

closed 2024-10-28 by uhlat

xiaohuirong, 2025-05-07:
I have an RTL8814AU USB adapter and encountered the same issue. After performing a bisect, I found that it broke after this commit: https://w1.fi/cgit/hostap/commit/?id=8056b79ff1e5d746a82c2ea12011e69e99c11d3d.

morrownr, 2025-05-18:
@xiaohuirong 

Did you report this hostpad?

xiaohuirong, 2025-05-19:
> [@xiaohuirong](https://github.com/xiaohuirong)
> 
> Did you report this hostpad?

I haven't reported this to hostapd, and I'm not sure whether it's an issue with hostapd. I'm asking here first to check.

morrownr, 2025-05-19:
@xiaohuirong 

I am assuming that you are using the driver in this repo. What needs to happen is I need to get you over to the new in-kernel driver to see if the same problem exist, then we can work it. I will not be doing any more work on the driver in this repo now that we have a good standards compliant driver available.

If you are using kernel 6.14, you can simply uninstall this driver and the in-kernel driver will take over on reboot. However, it you are on an older kernel, go to:

https://github.com/lwfinger/rtw88

And install the driver from there. It is the same code as what is in kernel 6.14 and later. That is where we developed the driver. If you see any problems, you can report the problem in that repo. See you there.

xiaohuirong, 2025-05-19:
My adapter is RTL8814AU, not RTL8812AU. Has this driver also been upstreamed? After uninstalling `rtl8814au-dkms` on kernel 6.14.2, it seems no related kernel module is being loaded.

xiaohuirong, 2025-05-19:
> > My adapter is RTL8814AU, not RTL8812AU.
> 
> I had assumed your adapter is based on the rtl8812au chip as this repo is for the rtl8812au chip.
> 
> The rtw88 repo I pointed you too has a new driver for the rtl8814au. The rtl8814au did not go upstream to 6.14 as it was not finished at the time. It started upstream with kernel 6.15 and more patches are coming with 6.16. Your best option for now is to use the rtw88 repo I provided the link to above as it does have the code for the rtl8814au. See you at the rtw88 repo.

Okay, thank you for your help. I'll try `rtw88` and give you feedback later.

xiaohuirong, 2025-05-19:
The issue has been resolved, but I encountered a problem with the rtw88 driver where the speed is halved. My previous module parameters for 8814au were:
```
options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=1
```
The hostapd configuration file was:
```
# https://github.com/morrownr/8814au/issues/20
interface=wlan1
bridge=br0

driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=NetPi

country_code=US

ieee80211d=1
ieee80211h=1

hw_mode=a

#channel=44
#vht_oper_centr_freq_seg0_idx=42
#Use the next line with channel 36 (36 + 6 = 42) band 2 - 5g
#vht_oper_centr_freq_seg0_idx=42
#channel=36
#Use the next line with channel 149 (149 + 6 = 155) band 2 - 5g
vht_oper_centr_freq_seg0_idx=155
channel=149

beacon_int=100
dtim_period=2
max_num_sta=32
macaddr_acl=0
rts_threshold=2347
fragm_threshold=2346

wpa_passphrase=xxxxx
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]

ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]

vht_oper_chwidth=1
```
Do I need to add some parameters to the rtw88 module?

morrownr, 2025-05-19:
> Do I need to add some parameters to the rtw88 module?

No, modern Linux drivers are far different than the old Realtek out-of-kernel drivers. Module parameters almost do not exist. The new drivers are very automatic. However, they require different setting in hostapd.conf.

I need to be out of the office for a while. I'll try to work that this afternoon.

xiaohuirong, 2025-05-19:
> No, modern Linux drivers are far different than the old Realtek out-of-kernel drivers. Module parameters almost do not exist. The new drivers are very automatic. However, they require different [...]

Alright, thank you so much for all the work you've put into these WiFi drivers and for your kind assistance.