morrownr/8812au-20210820 #42: ID 7392:b822 Edimax Technology Co., Ltd EW-7822ULC 802.11ac Wireless Adapter [Realtek RTL8812AU]

compsci282, 2024-11-12:
A slightly different ID for a variant of an Edimax adapter with Realtek RTL8812AU.

Entire lsusb output:
$ lsusb
Bus 001 Device 014: ID **7392:b822** Edimax Technology Co., Ltd EW-7822ULC 802.11ac Wireless Adapter [Realtek **RTL8812AU**]
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2024-11-12:
Hi @compsci282 

Thanks for the information. I just merged a patch to take care of this.

FYI: I'll go ahead and see about getting it pushed into the Linux kernel as well. The new rtl8812au and rtl8821/11au drivers have been approved and are queued up to go into Linux kernel 6.13. I think users are going to be pleased with the quality of the new drivers.

morrownr, 2024-11-12:
@compsci282 

On further checking, the ID you are reporting is not for this driver. It is for the rtl88x2bu driver and is already in place. I reverted the patch. Cheers.