morrownr/8812au-20210820 #43: [Bug]: AWUS036ACH not entering monitor mode on Raspberry Pi 5 using Raspberry Pi OS

neuterscooter, 2024-11-18:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

None

### uname

Linux faradaypi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### lsusb

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 2: phy1: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
rtl8812au/5.13.6-23, 6.6.51+rpt-rpi-2712, aarch64: installed
rtl8821au/5.12.5.2, 6.6.51+rpt-rpi-2712, aarch64: installed
```

### iw

```text
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 00:c0:ca:b4:09:b8
		type managed
		txpower -100.00 dBm
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 2c:cf:67:3d:fd:a4
		type managed
		channel 165 (5825 MHz), width: 20 MHz, center1: 5825 MHz
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

### install-driver.sh

```text
: ---------------------------
: install-driver.sh v20241003
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 8245584 (total system memory)
: 6.6.51+rpt-rpi-2712 (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was installed by dkms.
Module rtl8812au-5.13.6-23 for kernel 6.6.51+rpt-rpi-2712 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

8812au.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.6.51+rpt-rpi-2712/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...
Deleting module rtl8812au-5.13.6-23 completely from the DKMS tree.
Removing 8812au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8812au-5.13.6-23
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Sign command: /lib/modules/6.6.51+rpt-rpi-2712/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
kernelver=6.6.51+rpt-rpi-2712 ./dkms-make.sh......................
Signing module /var/lib/dkms/rtl8812au/5.13.6-23/build/8812au.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8812au.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.6.51+rpt-rpi-2712/updates/dkms/
depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates to this driver SHOULD be performed before major
      upgrades such as kernel 6.5 to 6.6.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 3 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] n
```

### What happened?

The AWUS036ACH adapter appears in iwconfig and lsusb and is able to connect to and see nearby networks in Network Manager. 

Running **aireplay-ng --test wlan1** yields:
**ioctl(SIOCSIWMODE) failed: Invalid argument
ioctl(SIOCSIWMODE) failed: Invalid argument
Error setting monitor mode on wlan1**

Running **hcxdumptool -i wlan1** yields:
**initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver: Invalid argument
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored**

I tried commenting out **blacklist rtw88_8812au** and **blacklist rtw_8812au** in the **8812au.conf** file, but that did not work. I also have a AWUS036ACS and AXML adapter that both enter monitor mode no problem, I don't know if those drivers cause any conflicts with the installed 8812au driver

morrownr, 2024-11-18:
Hi @neuterscooter 

Excellent report. I did not understand that your issue was with monitor mode. What you are experiencing is not a bug but rather that monitor mode is not supported in this driver. From the README.md:

```
Monitor mode is not supported. Linux users that want good support for monitor mode in a dual band or tri-band adapter should seek out USB WiFi adapters based on the mt7610u, mt7612u or mt7921au chipsets.

```

I made a decision about a year ago that any new out-of-kernel drivers that I bring up with not support monitor mode because Realtek simply does a bad job with monitor mode. I do not like to leave users hanging so there is a fallback plan. 

The new rtl8812au in-kernel driver was recently approved for merging into the Linux kernel. It is going into kernel 6.13 this week. In the meantime, you can install the repo we used to built it:

https://github.com/lwfinger/rtw88

Follow the directions at that repo as they are somewhat different than here. Once you have the repo installed, you will be running the code that is going into the kernel. This driver was approved by Realtek but the work was not done by Realtek. I would like a report on how monitor mode is working.

Let me know how it turns out.

morrownr, 2024-11-18:
And do not forget to fully uninstall/remove this driver.

rainst, 2025-02-10:
I came here because i was having the same issue, I just want to confirm that monitor mode works using the repo you linked.. 

Thank you!

morrownr, 2025-02-16:
@rainst 

? https://github.com/lwfinger/rtw88

Monitor mode works in the new driver. I have done basic testing but we need others to report results.

rainst, 2025-02-17:
I was having issues putting in monitor mode my AWUS036ACH (based on chipset RTL8812AU) on raspberry pi 5 with stock os and I got it to work after using the driver available here https://github.com/lwfinger/rtw88

@morrownr are you saying that it's supposed to work now? (despite the disclaimer in the README saying it is not supported)
I can try again if so

morrownr, 2025-02-17:
@rainst 

Let me clarify: The driver in this repo, 8812au-20210820, does not support monitor mode. There are no plans to support monitor mode. My efforts have been going into https://github.com/lwfinger/rtw88 . This rtw88 repo is being upstreamed into the linux kernel.

In fact, the new drivers (rtw88) for rtl8812au and rtl8821au started going into the kernel with kernel 6.13 and finished with 6.14. Once you are using kernel 6.14 or later, there is no need to find and compile a driver. Until you get to kernel 6.14, you can use https://github.com/lwfinger/rtw88 . A new driver for the rtl8814au chip is in the process of going into the kernel right now. If all goes well, there will be a new (rtw88) in-kernel driver for rtl8814au based adapters in kernel 6.15 if all goes well and 6.16 if there is some slippage in the timeline.

rainst, 2025-02-17:
thank you @morrownr, you have confirmed what I was able to figure out in the last couple of weeks!
Let me confirm again that the driver from the `rtw88` repo is working perfectly, so thank you (and other contributors) for your efforts!

I'm just a bit disappointed as a Debian user because it doesn't look like it will be included in Trixie, so I will keep compiling the driver from that repo for a while :)

morrownr, 2025-02-17:
> I'm just a bit disappointed as a Debian user because it doesn't look like it will be included in Trixie...

I am a Debian user also so I understand. On the other hand, the timing for Ubuntu 25.04 is perfect. 

The story: Dubhater, lwfinger and I started the project over one year ago. lwfinger passed away in May. Dubhater worked very hard but one big thing we ran into was the need to fix a LOT of things in rtw88. Patch after patch flowed into the kernel to get rtw88 into good enough shape to continue with the new drivers. We worked hard to get plenty of testers and helpers. I think rtw88 in kernel 6.14 is really good. A new driver for the rtl8814au chip is now going in and will hopefully be fully in kernel 6.15, if not, then kernel 6.16. As far as I can tell, that will give Linux users good drivers for all commonly used USB WiFi 5 chips.

One thing am trying to do is get the message out to Linux users to avoid Realtek WiFi 6 and 7 based adapters as most of those adapters are multi-state (windows driver onboard) and there is no usb support in rtw89. That makes for unhappy Linux users. But there is no reason to be unhappy as Mediatek is making great in-kernel drivers for WiFi 6 and 7.

https://github.com/morrownr/USB-WiFi

Scroll down and read items 1 and 2.

Cheers