morrownr/8812au-20210820 #44: [Bug]:

Timfedora, 2024-12-15:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

None

### uname

Linux fedora 6.11.10-300.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Nov 23 00:51:20 UTC 2024 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 8087:0a2a Intel Corp. Bluetooth wireless interface Bus 001 Device 003: ID 2232:1082 Silicon Motion 720p HD Camera Bus 001 Device 047: ID 05e3:0608 Genesys Logic, Inc. Hub Bus 001 Device 048: ID 0bda:881a Realtek Semiconductor Corp. RTL8812AU-VS 802.11a/b/g/n/ac 2T2R DB WLAN Adapter Bus 001 Device 049: ID 343c:0000 xxxxxxxx USB Type-C Digital AV Adapter Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

tim@fedora:~$ sudo rfkill list all 0: hci0: Bluetooth 	Soft blocked: yes 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 21: phy20: Wireless LAN 	Soft blocked: no 	Hard blocked: no tim@fedora:~$ 

### dkms

```text
rtl8812au/5.13.6-23, 6.11.10-300.fc41.x86_64, x86_64: installed
```

### iw

```text
sudo iw dev
phy#20
	Interface wlp0s20f0u2u1
		ifindex 23
		wdev 0x1400000001
		addr ce:ed:3e:ec:40:0b
		ssid Staples Store Guests
		type managed
		channel 48 (5240 MHz), width: 20 MHz, center1: 5240 MHz
		txpower 16.00 dBm
phy#0
	Unnamed/non-netdev interface
		wdev 0x4d
		addr f8:94:c2:aa:d7:7d
		type P2P-device
	Interface wlp1s0
		ifindex 2
		wdev 0x1
		addr fa:7a:db:80:80:b3
		type managed
		txpower 22.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	176	0	0	0	0	53602		176
```

### git status

```text
I downloaded the zip
```

### install-driver.sh

```text
sudo bash  install-driver.sh 
: ---------------------------
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 3894612 (total system memory)
: 6.11.10-300.fc41.x86_64 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 14.2.1 20240912 (Red Hat 14.2.1-3)
: dkms-3.1.1
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was installed by dkms.

Module rtl8812au-5.13.6-23 for kernel 6.11.10-300.fc41.x86_64 (x86_64):
Before uninstall, this module version was ACTIVE on this kernel.
Deleting /lib/modules/6.11.10-300.fc41.x86_64/extra/8812au.ko.xz
Running depmod....... done.

Deleting module rtl8812au-5.13.6-23 completely from the DKMS tree.
Removing 8812au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8812au-5.13.6-23
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Sign command: /lib/modules/6.11.10-300.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...................................... done.
Signing module /var/lib/dkms/rtl8812au/5.13.6-23/build/8812au.ko
Cleaning build area... done.
Compile time: 361.06 seconds
The driver was built by dkms successfully.
: ---------------------------

Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/8812au.ko.xz
Running depmod...... done.
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates to this driver SHOULD be performed before major
      upgrades such as kernel 6.5 to 6.6.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 3 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n]
```

### What happened?

Hi, first of all, thank you for creating this driver!

I’ve encountered an issue and I’m unsure if it’s related to the driver or a GNOME setting. When attempting to switch networks or connect/reconnect to a network that requires a sign-in but no password, it doesn’t seem to work as expected.

Here’s my system info:

OS: Fedora 41
Kernel: 6.11
GNOME Control Center Version: 47.2
GNOME Shell: 47.1

https://github.com/user-attachments/assets/70b4fb5a-c331-4555-8b00-0b8a461a3cf4

morrownr, 2024-12-15:
Hi @Timfedora 

> When attempting to switch networks or connect/reconnect to a network that requires a sign-in but no password, it doesn’t seem to work as expected.

I sometimes run into this situation with my laptop. I connect and then when I try to go to a site with my browser, a sign on page pops up and I do what they want and I go about my business. This problem doesn't really have anything to do with this driver unless I am missing something. You may think you are not connected but have you tried to go to a site?

FYI: A nice standards compliant in kernel driver just went into kernel 6.13 for your adapter. I'd say you can count on it working at full speed by kernel 6.14 so you have something to look forward to.

By the way, the Main Menu of this site is:

https://github.com/morrownr/USB-WiFi

Timfedora, 2024-12-22:
Hi,
I’ve figured out the hotspot login issue by entering neverssl.co, and that seems to have resolved it. However, I’m still experiencing issues with switching networks. I suspect this might be a problem with GNOME settings, as nmcli works fine without any issues.

Thanks for your help!

Timfedora, 2024-12-22:
> FYI: A nice standards compliant in kernel driver just went into kernel 6.13 for your adapter. I'd say you can count on it working at full speed by kernel 6.14 so you have something to look forward to.

That’s nice! I actually found my card in an old TV, soldered on a USB cable, and now I use it to get better reception when I’m in the car.  
[image]

morrownr, 2024-12-22:
That is really cool. I like to recycle things.