morrownr/8812au-20210820 #45: [Bug]:

Jussi73, 2025-01-02:
### Checklist

- [X] I am not using this driver from inside a Virtual Machine.
- [X] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [X] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux KodiBox 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

### lsusb

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 005: ID 0b05:17d2 ASUSTek Computer, Inc. USB-AC56 802.11a/b/g/n/ac Wireless Adapter [Realtek RTL8812AU] Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: hci0: Bluetooth         Soft blocked: no         Hard blocked: no 1: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no 2: phy1: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
rtl8821au/5.12.5.2, 6.6.62+rpt-rpi-v8, aarch64: installed
```

### iw

```text
phy#1
        Interface wlan1
                ifindex 4
                wdev 0x100000001
                addr 98:25:4a:d4:e0:67
                ssid TPLink
                type managed
                channel 2 (2417 MHz), width: 20 MHz, center1: 2417 MHz
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr e6:5f:01:c1:63:a6
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr e4:5f:01:c1:63:a6
                type managed
                channel 140 (5700 MHz), width: 20 MHz, center1: 5700 MHz
                txpower 31.00 dBm
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

### install-driver.sh

```text
: install-driver.sh v20240314
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 1891124 (total system memory)
: 6.6.62+rpt-rpi-v8 (kernel version)
: gcc-12 (version of gcc used to compile the kernel)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Module rtl8821au-5.12.5.2 for kernel 6.6.62+rpt-rpi-v8 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

8821au.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.6.62+rpt-rpi-v8/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....
Deleting module rtl8821au-5.12.5.2 completely from the DKMS tree.
Removing 8821au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821au-5.12.5.2
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821au-5.12.5.2
Creating symlink /var/lib/dkms/rtl8821au/5.12.5.2/source -> /usr/src/rtl8821au-5.12.5.2
The driver was added to dkms successfully.
: ---------------------------

Sign command: /lib/modules/6.6.62+rpt-rpi-v8/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
kernelver=6.6.62+rpt-rpi-v8 ./dkms-make.sh...........................................................
Signing module /var/lib/dkms/rtl8821au/5.12.5.2/build/8821au.ko
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8821au.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.6.62+rpt-rpi-v8/updates/dkms/
depmod....
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates to this driver SHOULD be performed before major
      upgrades such as kernel 6.5 to 6.6.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 3 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!
```

### What happened?

I have installed these drivers on a fresh Pi4 (64bit). All updates are performed according to instructions here. Compile and install ran  without errors. 

The AC56 device shows up if I run LSUSB, but it does not show up under WLAN devices. I have the onboard Pi4 wifi, and a crappy TPlink dongle in the Pi now and they both show up - see LSUSB output above. 

How do I get it to show up as a WLAN device that I can actually use?

Thanks!

morrownr, 2025-01-02:
Hi @Jussi73 

> How do I get it to show up as a WLAN device that I can actually use?

Install the correct driver. You have installed the rtl8821au driver. You need the rtl8812au driver:

https://github.com/morrownr/8812au-20210820

Jussi73, 2025-01-03:
Uninstalled RTL8821au driver using the uninstall script. Ran the process again as outlined above. Adapter is up and running good now. Thanks :-)

I dunno how I got 8821au installed. This was a fresh Pi, and I could have sworn that I ran everything as outlined in the readme, but apparently something went haywire there, but that was a pretty dumb mistake....

Anyway, thank you for answering and thank you for your work. Much appreciated - this solves a network issue I had in the simplest possible way with hardware I had laying around. 

Will close this thread - obviously not a bug.

closed 2025-01-03 by Jussi73

morrownr, 2025-01-03:
FYI: A new in-kernel driver for this chip was added with kernel 6.13. I'd say once are using kernel 6.14 or later, you won't need to find a driver anymore.