morrownr/8812au-20210820 #46: make fails opensuse leap 15.6, kernel 6.4.0-150600.23.30-default

mjopensuse, 2025-01-14:
issuing 'Make' fails at the following stage:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.4.0-150600.23.30-default/build M=/root/github/8812au-20210820  modules
make[1]: Entering directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
 CC [M]  /root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.o
/root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.c:10481:19: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
 .change_beacon = cfg80211_rtw_change_beacon,
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.c:10481:19: note: (near initialization for ‘rtw_cfg80211_ops.change_beacon’)
/root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.c:10553:15: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
 .tdls_mgmt = cfg80211_rtw_tdls_mgmt,
              ^~~~~~~~~~~~~~~~~~~~~~
/root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.c:10553:15: note: (near initialization for ‘rtw_cfg80211_ops.tdls_mgmt’)
cc1: warning: unrecognized command line option ‘-Wno-enum-conversion’
cc1: warning: unrecognized command line option ‘-Wno-stringop-overread’
cc1: warning: unrecognized command line option ‘-Wno-enum-int-mismatch’
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-6.4.0-150600.23.30/scripts/Makefile.build:252: /root/github/8812au-20210820/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [../../../linux-6.4.0-150600.23.30/Makefile:2064: /root/github/8812au-20210820] Error 2
make[1]: Leaving directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
make: *** [Makefile:2516: modules] Error 2

Any suggestions on how to solve?

mjopensuse, 2025-01-14:
Thanks for the quick reply! I tried the rtw88 you mentioned but it also does not compile for me. I would be interested in the patch. Seems my best shot it trying to get the driver from the current github repo to work. Commenting out the 2 lines of error mentioned in the issue does yield it compiles completely, although after modprobe 8812au it does not pick up the wifi dongle...

morrownr, 2025-01-15:
> I tried the rtw88 you mentioned but it also does not compile for me.

It may be the same issue. 

> Seems my best shot it trying to get the driver from the current github repo to work.

What I would like is if you could paste the errors you are seeing so I can make sure to get on the correct patch.

With the driver in this repo, you may need to do the compile manually to the best picture of the error:

$ make clean
$ make
once you copy and paste the part where the compilation fails, you can clean up again...
$ make clean

mjopensuse, 2025-01-16:
The error listed in my first post is the result of issuing make.

morrownr, 2025-01-16:
After investigating, I determined that I was wrong with my initial assessment. 

The main candidate right now is:

gcc version 7.5.0

I'm scratching my head while wondering how you have a fairly modern kernel combined with a VERY old compiler. There are several years difference in the age of the kernel and the compiler. Even one or two years difference can lead to problems but in this case, we are talking about several years. 

The README, in the Installation Steps section, has information about this topic in item 7. You can check what version your kernel was compiled with:

$ cat /proc/version

mjopensuse, 2025-01-16:
kernel is compiled with 7.5.0. I don't understand really why opensuse leap 15.6 has chosen to stay with such an old version. 

For the errors though seems the 6.4 kernel just has a different API for cfg80211 than what is used in your branch. 

Would there be some commit that I can try to drop back to -- basically some salient version up to the point where you worked against kernel 6.4 back in the day?

morrownr, 2025-01-16:
> Would there be some commit that I can try to drop back to -- basically some salient version up to the point where you worked against kernel 6.4 back in the day?

I do not know as it has been years since I tested anything here with gcc 7.5. I test with gcc 11, 12, 13 and 14 these days and was considering dropping 11. I do not remember problems like this when 6.4 was more commonly used but it is not an LTS kernel so it is not maintained unless the OpenSuSE folks backport patches.

I would prefer to pursue this problem with rtw88 as my plan with these Realtek wifi 5 out-of-kernel drivers is to discontinue ongoing maintenance with kernel 6.12 as kernels later than that should have pretty good quality in-kernel rtw88 drivers for Realtek wifi 5. Would you mind showing me where rtw88 blew up?

mjopensuse, 2025-01-16:
Sure thing, here is the error I get with rtw88:

linux-jai1:~/github/rtw88 # git status
On branch 8821au_8812au
Your branch is up to date with 'origin/8821au_8812au'.

nothing to commit, working tree clean
linux-jai1:~/github/rtw88 # make clean
linux-jai1:~/github/rtw88 # make
make -C /lib/modules/6.4.0-150600.23.30-default/build M=/root/github/rtw88 modules
make[1]: Entering directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
  CC [M]  /root/github/rtw88/main.o
In file included from /root/github/rtw88/main.c:7:0:
/root/github/rtw88/main.h:45:1: error: redefinition of ‘cfg80211_ssid_eq’
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-6.4.0-150600.23.30/include/net/mac80211.h:22:0,
                 from /root/github/rtw88/main.h:12,
                 from /root/github/rtw88/main.c:7:
/usr/src/linux-6.4.0-150600.23.30/include/net/cfg80211.h:7226:1: note: previous definition of ‘cfg80211_ssid_eq’ was here
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-6.4.0-150600.23.30/scripts/Makefile.build:252: /root/github/rtw88/main.o] Error 1
make[1]: *** [../../../linux-6.4.0-150600.23.30/Makefile:2064: /root/github/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
make: *** [Makefile:153: all] Error 2

morrownr, 2025-01-16:
> On branch 8821au_8812au 

I need that to say `On branch Master`

Now I understand the problem.

mjopensuse, 2025-01-17:
problem with Master: 

linux-jai1:~/github/rtw88 # git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
linux-jai1:~/github/rtw88 # make clean 
make -C /lib/modules/6.4.0-150600.23.30-default/build M=$PWD clean
make[1]: Entering directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
make[1]: Leaving directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
linux-jai1:~/github/rtw88 # make
make -j`nproc` -C /lib/modules/6.4.0-150600.23.30-default/build M=$PWD modules
make[1]: Entering directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
  CC [M]  /root/github/rtw88/main.o
  CC [M]  /root/github/rtw88/mac80211.o
In file included from /root/github/rtw88/main.c:7:0:
/root/github/rtw88/main.h:69:1: error: redefinition of ‘cfg80211_ssid_eq’
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
In file included from /root/github/rtw88/mac80211.c:5:0:
/root/github/rtw88/main.h:69:1: error: redefinition of ‘cfg80211_ssid_eq’
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-6.4.0-150600.23.30/include/net/mac80211.h:22:0,
                 from /root/github/rtw88/main.h:12,
                 from /root/github/rtw88/main.c:7:
/usr/src/linux-6.4.0-150600.23.30/include/net/cfg80211.h:7226:1: note: previous definition of ‘cfg80211_ssid_eq’ was here
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-6.4.0-150600.23.30/include/net/mac80211.h:22:0,
                 from /root/github/rtw88/main.h:12,
                 from /root/github/rtw88/mac80211.c:5:
/usr/src/linux-6.4.0-150600.23.30/include/net/cfg80211.h:7226:1: note: previous definition of ‘cfg80211_ssid_eq’ was here
 cfg80211_ssid_eq(struct cfg80211_ssid *a, struct cfg80211_ssid *b)
 ^~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-6.4.0-150600.23.30/scripts/Makefile.build:252: /root/github/rtw88/mac80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-6.4.0-150600.23.30/scripts/Makefile.build:252: /root/github/rtw88/main.o] Error 1
make[1]: *** [../../../linux-6.4.0-150600.23.30/Makefile:2064: /root/github/rtw88] Error 2
make[1]: Leaving directory '/usr/src/linux-6.4.0-150600.23.30-obj/x86_64/default'
make: *** [Makefile:183: all] Error 2

mjopensuse, 2025-01-18:
ok i got make to complete, by making sure this function is not redefined ( it does so if kernel < 6.8, since I am on 6.4 I bumped that down to get past the error). I seem to have the modules now 

linux-jai1:/lib/modules/6.4.0-150600.23.33-default/kernel/drivers/net/wireless/realtek/rtw88 # ls *.ko
rtw_8703b.ko   rtw_8723d.ko   rtw_8723x.ko   rtw_8814a.ko   rtw_8821au.ko  rtw_8821cs.ko  rtw_8822b.ko   rtw_8822ce.ko  rtw_8822cu.ko  rtw_pci.ko
rtw_8723cs.ko  rtw_8723ds.ko  rtw_8812a.ko   rtw_8814au.ko  rtw_8821ce.ko  rtw_8821cu.ko  rtw_8822bs.ko  rtw_8822c.ko   rtw_88xxa.ko   rtw_sdio.ko
rtw_8723de.ko  rtw_8723du.ko  rtw_8812au.ko  rtw_8821a.ko   rtw_8821c.ko   rtw_8822be.ko  rtw_8822bu.ko  rtw_8822cs.ko  rtw_core.ko    rtw_usb.ko

I am pretty sure I need rtw8812au. Device I am using is TP-Link Archer T2U Nano, has 8812au chipset. I do

modprobe rtw_8812au

then 

linux-jai1:/lib/modules/6.4.0-150600.23.33-default/kernel/drivers/net/wireless/realtek/rtw88 # lsmod | grep rtw
rtw_8812au             12288  0
rtw_8812a              53248  1 rtw_8812au
rtw_88xxa              45056  1 rtw_8812a
rtw_usb                36864  1 rtw_8812au
rtw_core              331776  3 rtw_88xxa,rtw_usb,rtw_8812a
mac80211             1527808  3 rtw_usb,rtw_core,ath9k
cfg80211             1290240  5 ath9k_common,rtw_core,ath9k,ath,mac80211
usbcore               434176  8 xhci_hcd,rtw_usb,rtsx_usb,uvcvideo,ath3k,btusb,xhci_pci,rtw_8812au

nothing showing up with iwconfig though 

Any suggestions? Let me show some dmesg output. I think only the last line is from the above modprobe action. Seems a retstart results in 8821au being loaded. That talks about verification failing. Maybe I just need to get past that, although I dont think I ned 8821a?

inux-jai1:/lib/modules/6.4.0-150600.23.33-default/kernel/drivers/net/wireless/realtek/rtw88 # dmesg | grep rtw
[   53.313623] rtw_core: loading out-of-tree module taints kernel.
[   53.313639] rtw_core: module verification failed: signature and/or required key missing - tainting kernel
[   53.682550] Modules linked in: rtw_8821au(OEn+) rtw_8821a(OEn) rtw_88xxa(OEn) rtsx_usb_ms(n) spi_intel(n) iTCO_wdt rtw_usb(OEn) intel_pmc_bxt videobuf2_common rtw_core(OEn) snd_timer iTCO_vendor_support btmtk mac80211 libarc4 kvm videodev bluetooth memstick cfg80211 snd realtek soundcore efi_pstore(n) mdio_devres irqbypass wmi_bmof mc pcspkr acer_wmi libphy intel_vbtn(n) mei_txe i2c_i801 ecdh_generic lpc_ich i2c_smbus i2c_designware_platform mei i2c_designware_core tiny_power_button(n) crc16 sparse_keymap soc_button_array(n) ccp rfkill_gpio rfkill fan ac dw_dmac(n) thermal button configfs fuse dmi_sysfs ip_tables x_tables rtsx_usb_sdmmc mmc_core rtsx_usb mfd_core crc32_pclmul i915 polyval_generic(n) gf128mul ahci libahci libata sd_mod xhci_pci scsi_dh_emc xhci_pci_renesas scsi_dh_rdac xhci_hcd scsi_dh_alua i2c_algo_bit t10_pi drm_buddy ttm ghash_clmulni_intel crc64_rocksoft_generic drm_display_helper usbcore cryptd sha512_ssse3 sha256_ssse3 crc64_rocksoft sg cec crc64 sha1_ssse3 scsi_mod rc_core video battery wmi
[   53.683150]  rtw_usb_probe+0x2c/0xae0 [rtw_usb cdc602713a9be3d76918044ef329399d280ab7ac]
[   53.683337]  ? __pfx_rtw_8821au_driver_init+0x10/0x10 [rtw_8821au 1c0a4ca9011d4fe8433d0f74aeb5cc654014d00a]
[   53.683505] rtw_8821au: probe of 1-2.2:1.0 failed with error -12
[   53.683568] usbcore: registered new interface driver rtw_8821au
[  279.098887] usbcore: deregistering interface driver rtw_8821au
[  367.203177] usbcore: registered new interface driver rtw_8812au
[  453.086402] Modules linked in: rtw_8821au(OEn+) rtw_8821a(OEn) rtw_8812au(OEn) rtw_8812a(OEn) rtw_88xxa(OEn) rfcomm qrtr(n) xt_LOG nf_log_syslog xt_limit xt_comment cmac algif_hash xt_pkttype algif_skcipher af_alg xt_tcpudp bnep ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 iptable_filter bpfilter ip6table_mangle nf_conntrack_netbios_ns nf_conntrack_broadcast xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6table_filter ip6_tables msr ext4 mbcache jbd2 nls_iso8859_1 nls_cp437 vfat fat snd_hda_codec_hdmi intel_rapl_msr intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic intel_soc_dts_thermal(n) intel_soc_dts_iosf(n) snd_hda_intel intel_powerclamp snd_intel_dspcfg spi_nor snd_intel_sdw_acpi hci_uart uvcvideo snd_hda_codec ath3k coretemp btusb btqca mtd videobuf2_vmalloc snd_hda_core uvc btrtl atomisp_lm3554(Cn) snd_hwdep joydev videobuf2_memops kvm_intel btintel r8169 snd_pcm videobuf2_v4l2 atomisp_gmin_platform(Cn) spi_intel_platform(n) btbcm mei_hdcp(n) hid_multitouch hid_generic mei_pxp
[  453.086518]  at24 regmap_i2c rtsx_usb_ms(n) spi_intel(n) iTCO_wdt rtw_usb(OEn) intel_pmc_bxt videobuf2_common rtw_core(OEn) snd_timer iTCO_vendor_support btmtk mac80211 libarc4 kvm videodev bluetooth memstick cfg80211 snd realtek soundcore efi_pstore(n) mdio_devres irqbypass wmi_bmof mc pcspkr acer_wmi libphy intel_vbtn(n) mei_txe i2c_i801 ecdh_generic lpc_ich i2c_smbus i2c_designware_platform mei i2c_designware_core tiny_power_button(n) crc16 sparse_keymap soc_button_array(n) ccp rfkill_gpio rfkill fan ac dw_dmac(n) thermal button configfs fuse dmi_sysfs ip_tables x_tables rtsx_usb_sdmmc mmc_core rtsx_usb mfd_core crc32_pclmul i915 polyval_generic(n) gf128mul ahci libahci libata sd_mod xhci_pci scsi_dh_emc xhci_pci_renesas scsi_dh_rdac xhci_hcd scsi_dh_alua i2c_algo_bit t10_pi drm_buddy ttm ghash_clmulni_intel crc64_rocksoft_generic drm_display_helper usbcore cryptd sha512_ssse3 sha256_ssse3 crc64_rocksoft sg cec [... cut ...]
[  453.087120]  rtw_usb_probe+0x2c/0xae0 [rtw_usb cdc602713a9be3d76918044ef329399d280ab7ac]
[  453.087307]  ? __pfx_rtw_8821au_driver_init+0x10/0x10 [rtw_8821au 1c0a4ca9011d4fe8433d0f74aeb5cc654014d00a]
[  453.087485] rtw_8821au: probe of 1-2.3:1.0 failed with error -12
[  453.087552] usbcore: registered new interface driver rtw_8821au
[  608.581368] usbcore: deregistering interface driver rtw_8812au
[  608.601131] usbcore: deregistering interface driver rtw_8821au
[ 1119.290083] usbcore: registered new interface driver rtw_8812au

mjopensuse, 2025-01-18:
[dmesg.log](https://github.com/user-attachments/files/18466070/dmesg.log)

morrownr, 2025-01-18:
> Device I am using is TP-Link Archer T2U Nano, has 8812au chipset.

A quick search seems to indicate that adapter uses the rtl8821/11au chip and the rtw_8821au driver in this repo. What is the usb ID?

$ lsusb

mjopensuse, 2025-01-20:
Bus 001 Device 008: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]

mjopensuse, 2025-01-20:
Meanwhile success on my end getting the device to run. Just posting here in case some other Leap 15.6 user runs into the same issue. 

Used https://github.com/aircrack-ng/rtl8812au/issues/1216, and following their suggestion to drop back to commit 

https://github.com/aircrack-ng/rtl8812au/commit/63cf0b4584aa8878b0fe8ab38017f31c319bde3d

In addition some simple fixes to get this branch to compile as follows: 
iff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index c1f20e9..ae7e131 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -5283,7 +5283,7 @@ exit:
        return ret;
 }
 
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 7, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0))
 static int cfg80211_rtw_change_beacon(struct wiphy *wiphy, struct net_device *ndev,
                struct cfg80211_ap_update *params)
 {
diff --git a/os_dep/linux/wifi_regd.c b/os_dep/linux/wifi_regd.c
index 45c5bd1..0d964dc 100644
--- a/os_dep/linux/wifi_regd.c
+++ b/os_dep/linux/wifi_regd.c
@@ -405,7 +405,7 @@ static void _rtw_regd_init_wiphy(struct rtw_regulatory *reg, struct wiphy *wiphy
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 19, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 1, 39)) \
        || (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 3, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 3, 13)) \
        || (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 4, 0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(6, 4, 4))
-       wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
+       wiphy->regulatory_flags |= REGULATORY_COUNTRY_IE_IGNORE;
 #endif

KottV, 2025-01-30:
@mjopensuse https://build.opensuse.org/package/show/hardware/rtl8812au

mjopensuse, 2025-02-11:
Let's close issue. Enough pointers on how to work around the issue for leap 15.6 users.

closed 2025-02-11 by mjopensuse

morrownr, 2025-02-11:
@mjopensuse 

You have another option:

https://github.com/lwfinger/rtw88

That is where we have been developing the new standards compliant drivers for rtl8812au and rtl8821/11au. They just went into kernels 6.13 and 6.14. The above repo gives you the same code and will allow you to test and report any problems... which will help everyone with these adapters. The drivers are quite good.

Cheers