morrownr/8812au-20210820 #47: Tested working on Ubuntu 20.04 6.8.0-51

Taikakim, 2025-01-27:
Tested on Linux kim-HP-EliteDesk-800-G2-DM-65W 6.8.0-51-lowlatency #52.1-Ubuntu SMP PREEMPT_DYNAMIC Sat Dec  7 23:21:04 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

Working! This is so great, I spent some  time looking for a driver which works. My device is the TP-Link Archer T4U, AC1300

Taikakim, 2025-02-11:
Ok, that is great news! I was wondering why the drivers are not in the kernel since they seem to work quite universally.

morrownr, 2025-02-12:
> I was wondering why the drivers are not in the kernel since they seem to work quite universally.

This driver is not what went into the kernel. This driver and all other Realtek out-of-kernel drivers are not Linux Standards compliant and will not be allowed into the kernel. The new driver is written by a Linux community dev with me helping. The only company that is supporting their USB chips with good quality in-kernel drivers is Mediatek.

Like I said above:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 may provide information that is useful.