morrownr/8812au-20210820 #48: Question: What is the theoretical file transfer speed?

holefrog, 2025-02-02:
Thank you for your time and effort in providing such a great driver. 

I have a very simple yet confusing question: What is the theoretical file transfer speed?

My Setup is:
Router: ASUS RT-AC5300, 5GHz, Channel 153, width: 80M.
NAS: QNAP 453BMini.
PC 1: Thinkpad X230, Ubuntu 24.04.1 LTS, AX210 Mini PCIe WIFI card, SSD hard disk. 
PC 2: Macbook Pro 2016 (A1708), Ubuntu 24.04.1 LTS, AC1200 USB WIFI card (8812AU) with two Antennas, SSD hard disk.

PC1 and PC2 both have a bit rate of 867 Mb/s. 
PC1 can achieve a transfer speed of 30 MB/s from the NAS. Although this is not an ideal speed, the X230 is quite an old model. However, PC2 achieves less than 15 MB/s while transferring the same file from the NAS.

Are there any settings I can modify to get better speed?

Here is information of  Thinkpad X230:
```
00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev 04)
00:16.3 Serial controller: Intel Corporation 7 Series/C210 Series Chipset Family KT Controller (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (Lewisville) (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C216 Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.1 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 2 (rev c4)
00:1c.2 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation QM77 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C216 Chipset Family SMBus Controller (rev 04)
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
03:00.0 Network controller: Intel Corporation Wi-Fi 6E(802.11ax) AX210/AX1675* 2x2 [Typhoon Peak] (rev 1a)

wlp3s0    IEEE 802.11  ESSID:"5300_5G-1"  
          Mode:Managed  Frequency:5.2 GHz  Access Point: 18:31:BF:A9:C8:94   
          Bit Rate=866.7 Mb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=68/70  Signal level=-42 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:29   Missed beacon:0

```

Here is information of Macbook Pro 2016:
```
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 014: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 015: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 016: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 017: ID 17ef:6044 Lenovo ThinkPad Laser Mouse
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

iwconfig 
lo        no wireless extensions.

wlx0013eff211e2  IEEE 802.11AC  ESSID:"5300_5G-2"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.765 GHz  Access Point: 18:31:BF:A9:C8:98   
          Bit Rate:867 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=59/100  Signal level=58/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```

morrownr, 2025-02-02:
Hi @holefrog 

> What is the theoretical file transfer speed?

With an AC1200 adapter with a 867 Mbps link speed, you can see around 430 Mbps transfer speed.

> Link Quality=68/70  Signal level=-42 dBm  

This connection is good.

> Link Quality=59/100  Signal level=58/100

This connection is not so good so if all else is equal, you should see better transfer speed with the first system.

holefrog, 2025-02-02:
> Hi [@holefrog](https://github.com/holefrog)
>
>
> With an AC1200 adapter with a 867 Mbps link speed, you can see around 430 Mbps transfer speed.
>
>
> This connection is good.
>
>
> This connection is not so [...]

Yes, even 430Mbps is arond 30-40MB/s, my issue is 15MB/s, and I am within 5 meters to the router. Both PC1 and PC2 are at the same location. 
Any reason can cause the Link Quality?

holefrog, 2025-02-03:
> > Any reason can cause the Link Quality?
> 
> I can think of a few things that could cause this. How about you start with checking the USB mode:
> 
> $ lsusb -t
> 
> 5000=USB3 480= USB2

You are quite right. 
```
        |__ Port 003: Dev 004, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
```
It is a USB2.0 interface, althhough it say AC-1200Mbps. Therefor, the copy speed should around 20-40MB/s.

morrownr, 2025-02-03:
> Driver=rtl8812au, 480M

This is the cause of the slower speed. What is the cause of the adapter being in USB2 mode? There could be various reasons.

- The USB port could be a USB2 port
- You are using an extension cable or powered hub that is USB2
- You do not have the driver set to switch to USB3

holefrog, 2025-02-03:
> This is the cause of the slower speed. What is the cause of the adapter being in USB2 mode? There could be various reasons.
>
>     * The USB port could be a USB2 port
>
>     * You are using an extension [...]

I tried two methods to connect it to the Thunderbolt port.   Both give me a USB 2.0 connection. Maybe this 8812AU is a USB 2.0 product?

1. Using a Google TypeC to USBA convert to connect.
2. Using a Lenovo TypeC to USBA splitter, which provides 3 USB 3.0 Ports and a Gigabyte Ethernet port.

```
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 014: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 015: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 016: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 017: ID 17ef:6044 Lenovo ThinkPad Laser Mouse
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

morrownr, 2025-02-03:
> Maybe this 8812AU is a USB 2.0 product?

Some usb adapters that also have bluetooth support are limited to USB2 but that is not the case with adapters based on rtl8812au chips as they are wifi-only chips. You adapter is capable of USB3.

Go into the driver directory and open a terminal. Run:

$ sudo edit-options.sh

That will open a file. Use your mouse and copy the line that starts with `options`. Paste what you get. Ctrl + x will close the file.

holefrog, 2025-02-04:
> Some usb adapters that also have bluetooth support are limited to USB2 but that is not the case with adapters based on rtl8812au chips as they are wifi-only chips. You adapter is capable of USB3. [...]

Only one line begins with options
```
options 8812au rtw_switch_usb_mode=0 rtw_led_ctrl=1

```

morrownr, 2025-02-05:
> Only one line begins with options

That is the way it should be.

> rtw_switch_usb_mode=0

Hmmm...there is the problem. Change that parameter to:

rtw_switch_usb_mode=1

and save the file. If using nano, Ctrl + O, Ctrl + X

Reboot. Tell me what you see now.

holefrog, 2025-02-05:
> > Only one line begins with options
> 
> That is the way it should be.
> 
> > rtw_switch_usb_mode=0
> 
> Hmmm...there is the problem. Change that parameter to:
> 
> rtw_switch_usb_mode=1
> 
> and save the file. If using nano, Ctrl + O, Ctrl + X
> 
> Reboot. Tell me what you see now.

Thank you for all the replies.

I installed Windows on my MBP and compared the speeds of the internal BCM4350 and the external 8812. The BCM reached 70MB/s, while the 8812 only reached 30MB/s. These results suggest that my 8812 card is using a USB 2.0 interface, even though its port is blue like a USB 3.0 port.

I then tried using only the BCM4350 on Ubuntu, but the connection was very slow and unstable. The maximum connection speed I achieved was 208Mbps, which rapidly decreased to 108, 54, 27... and sometimes only 4Mbps. That's why I tried using the 8812.

It seems that the only viable option is to use Windows on my MBP.

morrownr, 2025-02-05:
> These results suggest that my 8812 card is using a USB 2.0 interface, even though its port is blue like a USB 3.0 port.

Does not matter what the color of the port is. I showed you how to check and it showed USB2. I then showed you what the problem is and how to fix it... but it appears that you did not fix it.

Your adapter will have a nice in-kernel driver starting with kernel 6.14.

holefrog, 2025-02-06:
> Does not matter what the color of the port is. I showed you how to check and it showed USB2. I then showed you what the problem is and how to fix it... but it appears that you did not fix it.
>
> Your [...]

I checked the info using USBView. I have no idea why the manufacturer  put 0x200 in bcdUSB. 

```
    ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x0BDA (Realtek Semiconductor Corp.)
idProduct                : 0x8812
bcdDevice                : 0x0000
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "Realtek"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "802.11n NIC"
iSerialNumber            : 0x03 (String Descriptor 3)
 Language 0x0409         : "123456"
bNumConfigurations       : 0x01 (1 Configuration)
Data (HexDump)           : 12 01 00 02 00 00 00 40 DA 0B 12 88 00 00 01 02   .......@........
```