morrownr/8812au-20210820 #49: [Bug]: compiling/building the driver fails on Arch 6.13.1-arch1-1 at os_dep/linux/os_intfs.o: error: expected ‘,’ or ‘;’ before ‘VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver’

BusterLundgaard, 2025-02-04:
### Checklist

- [x] I am not using this driver from inside a Virtual Machine.
- [x] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [x] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux buster 6.13.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 02 Feb 2025 01:02:29 +0000 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 003: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) Bus 001 Device 005: ID 2357:0101 TP-Link RTL8812AU Archer T4U 802.11ac Bus 001 Device 006: ID 1a2c:497a China Resource Semico Co., Ltd AFFFA0319 Bus 001 Device 008: ID 2717:ff80 Xiaomi Inc. Mi/Redmi series (RNDIS) Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no

### dkms

```text
(none, i have nothing installed)
```

### iw

```text
(none)
```

### git status

```text
Your branch is up to date with 'origin/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.8812au.mod.cmd
	8812au.mod
	core/.rtw_ap.o.cmd
	core/.rtw_beamforming.o.cmd
	core/.rtw_br_ext.o.cmd
	core/.rtw_btcoex.o.cmd
	core/.rtw_btcoex_wifionly.o.cmd
	core/.rtw_chplan.o.cmd
	core/.rtw_cmd.o.cmd
	core/.rtw_debug.o.cmd
	core/.rtw_ft.o.cmd
	core/.rtw_ieee80211.o.cmd
	core/.rtw_io.o.cmd
	core/.rtw_ioctl_query.o.cmd
	core/.rtw_ioctl_set.o.cmd
	core/.rtw_iol.o.cmd
	core/.rtw_mbo.o.cmd
	core/.rtw_mi.o.cmd
	core/.rtw_mlme.o.cmd
	core/.rtw_mlme_ext.o.cmd
	core/.rtw_odm.o.cmd
	core/.rtw_p2p.o.cmd
	core/.rtw_pwrctrl.o.cmd
	core/.rtw_recv.o.cmd
	core/.rtw_rf.o.cmd
	core/.rtw_rm.o.cmd
	core/.rtw_rm_fsm.o.cmd
	core/.rtw_rm_util.o.cmd
	core/.rtw_roch.o.cmd
[... 238 lines cut ...]
                 from os_dep/linux/os_intfs.c:17:
os_dep/linux/os_intfs.c:35:26: error: expected ‘,’ or ‘;’ before ‘VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver’
   35 |         MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.13.1-arch1-1/build/include/linux/moduleparam.h:26:61: note: in definition of macro ‘__MODULE_INFO’
   26 |                 = __MODULE_INFO_PREFIX __stringify(tag) "=" info
      |                                                             ^~~~
/usr/lib/modules/6.13.1-arch1-1/build/include/linux/module.h:299:33: note: in expansion of macro ‘MODULE_INFO’
  299 | #define MODULE_IMPORT_NS(ns)    MODULE_INFO(import_ns, ns)
      |                                 ^~~~~~~~~~~
os_dep/linux/os_intfs.c:35:9: note: in expansion of macro ‘MODULE_IMPORT_NS’
   35 |         MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
      |         ^~~~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.13.1-arch1-1/build/scripts/Makefile.build:194: os_dep/linux/os_intfs.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.1-arch1-1/build/Makefile:1989: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.1-arch1-1/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/buster/src/8812au-20210820'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.1-arch1-1/build'
make: *** [Makefile:2516: modules] Error 2
```

This seems like a bug no? But maybe i messed something up by trying to uninstall directly. I don't quite know where to go from here though.

Kind regards, and thanks for the amazing work!

morrownr, 2025-02-04:
Hi @BusterLundgaard 

> This seems like a bug no?

We can call it a bug. It is just the periodic updates to kernel APIs that require maintenance to this driver to continue to work. You are out on the bleeding edge I see.

I have merged a patch that should take care of the problem but I need you to test and report back.

To apply the changes:

$ git pull

$ sudo sh install-driver.sh

FYI: When you get to kernel 6.14, a new in-kernel driver will be there so your adapter should be plug and play. Hopefully your distro adds the firmware and turns on the compilation of the driver. If you have problems, let me know.

BusterLundgaard, 2025-02-04:
It works! Thanks so much!
Also, the plug and play sounds like a great thing to look forward to haha.

BusterLundgaard, 2025-06-13:
> Thanks for the report.
> 
> > Also, the plug and play sounds like a great thing to look forward to haha.
> 
> It is a very good thing. The new driver is much better than this one.

Hi. I am unfortunately having issues once again with compiling/building the driver after a new update. Before I make an issue though, I just wanna ask how one uses this new plug n play driver? I've updated everything on arch, but I still seem to need this unofficial one. 

Plug n play means: "Just put in your wifi adapter and the drivers should already be there in the kernel, no?"

BusterLundgaard, 2025-06-14:
Hi. I removed the driver with the uninstall script, rebooted and did a full system update with Pacman. Unfortunately, it seems my wifi adapter isn't being recognized. Running

`ip addr`

with the adapter in gives me the same as running it without the adapter in.

Regarding this custom driver, i get the following build error:

`
[buster@buster 8812au-20210820]$ cat /var/lib/dkms/rtl8812au/5.13.6-23/build/make.log
DKMS (dkms-3.2.1) make.log for rtl8812au/5.13.6-23 for kernel 6.15.2-arch1-1 (x86_64)
Sat Jun 14 12:19:06 PM CEST 2025

Building module(s)
# command: kernelver=6.15.2-arch1-1 ./dkms-make.sh
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.15.2-arch1-1/build M=/var/lib/dkms/rtl8812au/5.13.6-23/build  modules
make[1]: Entering directory '/usr/lib/modules/6.15.2-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/rtl8812au/5.13.6-23/build'
  CC [M]  core/rtw_cmd.o
  CC [M]  core/rtw_security.o
  CC [M]  core/rtw_debug.o
  CC [M]  core/rtw_io.o
  CC [M]  core/rtw_ioctl_query.o
  CC [M]  core/rtw_ioctl_set.o
core/rtw_debug.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
core/rtw_security.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
  CC [M]  core/rtw_ieee80211.o
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_debug.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_security.o] Error 1
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_cmd.o] Error 1
core/rtw_io.c:47:10: fatal error: drv_types.h: No such file or directory
   47 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_io.o] Error 1
core/rtw_ioctl_query.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_ioctl_query.o] Error 1
core/rtw_ioctl_set.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_ioctl_set.o] Error 1
core/rtw_ieee80211.c:20:10: fatal error: drv_types.h: No such file or directory
   20 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.15.2-arch1-1/build/scripts/Makefile.build:203: core/rtw_ieee80211.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.2-arch1-1/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.2-arch1-1/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/rtl8812au/5.13.6-23/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.2-arch1-1/build'
make: *** [Makefile:2516: modules] Error 2

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------
`

Kind regards.

morrownr, 2025-06-15:
I just tried to duplicate the error here and was unable to do so. This system has Ubuntu 25.04 with kernel 6.14.

Let's concentrate on the new driver please.

Please post the results of the following command:

$ sudo ls -l /lib/firmware/rtw88

BusterLundgaard, 2025-06-15:
```
> sudo ls -l /lib/firmware/rtw88
total 412
-rw-r--r-- 1 root root   610 May 22 20:06 README.zst
-rw-r--r-- 1 root root 13117 May 22 20:06 rtw8703b_fw.bin.zst
-rw-r--r-- 1 root root 14647 May 22 20:06 rtw8703b_wow_fw.bin.zst
-rw-r--r-- 1 root root 17918 May 22 20:06 rtw8723d_fw.bin.zst
-rw-r--r-- 1 root root 16756 May 22 20:06 rtw8812a_fw.bin.zst
-rw-r--r-- 1 root root 36290 May 22 20:06 rtw8814a_fw.bin.zst
-rw-r--r-- 1 root root 19570 May 22 20:06 rtw8821a_fw.bin.zst
-rw-r--r-- 1 root root 53954 May 22 20:06 rtw8821c_fw.bin.zst
-rw-r--r-- 1 root root 77712 May 22 20:06 rtw8822b_fw.bin.zst
-rw-r--r-- 1 root root 85925 May 22 20:06 rtw8822c_fw.bin.zst
-rw-r--r-- 1 root root 64058 May 22 20:06 rtw8822c_wow_fw.bin.zst
```