morrownr/8812au-20210820 #50: Could you add Edimax AC600 USB 7392:a812

SpatMan05, 2025-02-27:
I have this USB Wifi adapter:
Bus 001 Device 002: ID 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB

Link to device on internet:
[](https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_ac600_dual-band/ew-7811utc/)

Could you add this to the driver (I think it is the good chipset)
Thanks in advance!

morrownr, 2025-02-28:
Hi @SpatMan05 

I'm pretty sure that this is not the correct driver for your adapter. 

The correct driver of this type is:

https://github.com/morrownr/8821au-20210708

I am, however, recommended a newer driver that we have been working on over the last year:

https://github.com/lwfinger/rtw88

The driver from that repo has gone into kernel 6.14 so once you are using kernel 6.14 or later, your adapter should be plug and play.   Ubuntu 25.04, to be released in April, is an example of a distro that will have the driver available soon. In the meantime, you can install the rtw88 driver at the repo above.

Let me know how it goes.

SpatMan05, 2025-02-28:
Hello @morrownr 
I directly tried rtw88 driver and it works perfectly!

I'm on Ubuntu 22.04 with kernel 6.8.0-52-generic
Big thanks for your quick answer!

closed 2025-02-28 by SpatMan05