morrownr/8812au-20210820 #51: [Bug]: Not working on Pi5 with latest ver

Deadangel368, 2025-04-07:
### Checklist

- [x] I am not using this driver from inside a Virtual Machine.
- [x] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [x] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

None

### uname

Linux kali-raspberry-pi5 6.6.63-v8+ #1 SMP PREEMPT Tue Dec  3 18:51:49 UTC 2024 aarch64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 1ea7:0066 SHARKOON Technologies GmbH [Mediatrack Edge Mini Keyboard] Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter Bus 003 Device 003: ID c0f4:03f5 Usb KeyBoard Usb KeyBoard Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

0: hci0: Bluetooth         Soft blocked: no         Hard blocked: no 1: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
realtek-rtl8814au/5.8.5.1~git20240527.d8208c8: added
rtl8812au/5.13.6-23: added
rtl8852bu/1.19.14-127, 6.6.63-v8+, aarch64: installed
```

### iw

```text
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr 42:e5:0e:be:fb:e8
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 2c:cf:67:73:d1:10
                ssid Optus_DD3EF3_5GHz
                type managed
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 31.00 dBm
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

### install-driver.sh

```text
---------------------------
: install-driver.sh v20241003
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 4083512 (total system memory)
: 6.6.63-v8+ (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Debian 14.2.0-19) 14.2.0
: dkms-3.1.5
: This system doesn't support Secure Boot
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was added to dkms.
Deleting module rtl8812au/5.13.6-23 completely from the DKMS tree.
Removing 8812au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8812au-5.13.6-23
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

The kernel is built without module signing facility, modules won't be signed

Cleaning build area... done.
Building module(s)........................(bad exit status: 2)
Failed command:
kernelver=6.6.63-v8+ ./dkms-make.sh

Error! Bad return status for module build on kernel: 6.6.63-v8+ (aarch64)
Consult /var/lib/dkms/rtl8812au/5.13.6-23/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

### What happened?

Installation not working with current version (6.6.63-v8+

morrownr, 2025-04-07:
Hi @Deadangel368 

I am going to offer you a couple of alternative solutions  as Kali for arm64 has been problematic and I do not have time to look at it this week.

Best alternative:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for my github site. I recommend reading menu item 1. Then proceed on to menu item 2 that contains The Plug and Play List as well as a list of adapters that are recommended for Kali. Buying adapters that are well known to work with Kali is a good idea.

Another alternative:

https://github.com/lwfinger/rtw88

This is a repo that we use to work on and test drivers that are intended to go upstream into Linux Mainline. In fact, the new driver for your rtl8812au based adapter is in kernel 6.14 and the new driver for your rtl8814au based adapter is in kernel 6.15. Since you are using a kernel earlier than those, you can install from the above repo.

Remember to first follow the instructions and remove any driver for the rtl8812au and rtl8814au based adapters before installing rtw88 as shown in the above link.

Deadangel368, 2025-04-07:
Thank you for getting back to me, as this is my 3rd adaptor, I'll try the alternative repo before anything else, I am happy to wait till you have the time to look into it for this repo if you do, as I have faith in you and prefer your code, but if forced to later I'll look into getting one of the reccomended adaptors, thank you for your time and patience

morrownr, 2025-04-07:
> as I have faith in you and prefer your code,

I work on the rtw88 "alternative" repo as well as this one. rtw88 is designed to be Linux Standards compliant (mac80211) so that the code will be accepted into the Linux kernel. I am planning to put this repo into a freeze with kernel 6.14 becoming available as the new rtl8812au driver is in kernel 6.14.

VanishingTacos, 2025-06-01:
rtw88 helped get my WUSB6300 working again on my Raspberry Pi! Thank you! <3

morrownr, 2025-06-01:
@VanishingTacos 

Glad the new driver helped. Keep in mind that once your distro is on kernel 6.14 or later, you should not need to install a driver and your adapter will be plug and play. The driver was accepted into the kernel.

If you see any problems, please report them to https://github.com/lwfinger/rtw88