morrownr/8812au-20210820 #54: [Bug]: Kernel oops on Ubuntu 24.04

lgleznah, 2025-09-04:
### Checklist

- [x] I am not using this driver from inside a Virtual Machine.
- [x] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [x] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux devuser 6.14.0-29-generic #29~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Aug 14 16:52:50 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 046d:c077 Logitech, Inc. Mouse Bus 001 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120 Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 002 Device 004: ID 2357:010e TP-Link Archer T4UH v2 [Realtek RTL8812AU]

### rfkill

2: phy2: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
rtl8812au/5.13.6-23, 6.14.0-28-generic, x86_64: installed
rtl8812au/5.13.6-23, 6.14.0-29-generic, x86_64: installed
rtl88x2bu/5.8.7.1, 6.14.0-28-generic, x86_64: installed
rtl88x2bu/5.8.7.1, 6.14.0-29-generic, x86_64: installed
```

### iw

```text
phy#2
        Interface wlx9c5322bc8466
                ifindex 5
                wdev 0x200000001
                addr 9c:53:22:bc:84:66
                ssid e2f
                type managed
                channel 48 (5240 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 16.00 dBm
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

### install-driver.sh

```text
: ---------------------------
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 15736024 (total system memory)
: 6.14.0-29-generic (kernel version)
: gcc-13 (version of gcc used to compile the kernel)
: gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
: dkms-3.0.11
: SecureBoot disabled
: ---------------------------

[... 125 lines cut ...]
[223527.001424] [      C6] Code: 23 f5 fe e8 58 ef ff ff 8b 53 04 49 89 c7 0f 1f 44 00 00 31 ff e8 06 34 f3 fe 80 7d d0 00 0f 85 22 02 00 00 fb 0f 1f 44 00 00 <45> 85 f6 0f 88 f2 01 00 00 4d 63 ee 49 83 fd 0a 0f 83 d8 04 00 00
[223527.001784] [      C6] RSP: 0018:ffffae2680127e00 EFLAGS: 00000246
[223527.002144] [      C6] RAX: 0000000000000000 RBX: ffff93d32ed43688 RCX: 0000000000000000
[223527.002510] [      C6] RDX: 0000000000000006 RSI: 0000000000000000 RDI: 0000000000000000
[223527.002873] [      C6] RBP: ffffae2680127e50 R08: 0000000000000000 R09: 0000000000000000
[223527.003234] [      C6] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff9a302160
[223527.003594] [      C6] R13: 0000000000000006 R14: 0000000000000006 R15: 0000cb4beec58dcc
[223527.003954] [      C6]  ? tick_nohz_stop_tick+0x64/0x280
[223527.004309] [      C6]  ? finish_task_switch.isra.0+0x9c/0x310
[223527.004665] [      C6]  cpuidle_enter+0x2e/0x50
[223527.005016] [      C6]  call_cpuidle+0x22/0x60
[223527.005362] [      C6]  cpuidle_idle_call+0x119/0x190
[223527.005712] [      C6]  do_idle+0x7f/0xe0
[223527.006056] [      C6]  cpu_startup_entry+0x29/0x30
[223527.006395] [      C6]  start_secondary+0x129/0x160
[223527.006780] [      C6]  common_startup_64+0x13e/0x141
[223527.007121] [      C6]  </TASK>
[223527.007495] [      C6] Modules linked in: tls snd_seq_dummy snd_hrtimer cmac nls_utf8 cifs cifs_arc4 nls_ucs2_utils cifs_md4 netfs qrtr binfmt_misc intel_rapl_msr intel_rapl_common snd_soc_avs nls_iso8859_1 intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp coretemp snd_soc_hda_codec kvm_intel snd_hda_ext_core snd_soc_core snd_hda_codec_hdmi kvm snd_hda_codec_realtek irqbypass rapl intel_cstate snd_compress ac97_bus snd_pcm_dmaengine snd_hda_codec_generic snd_hda_scodec_component snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec 8812au(OE) snd_hda_core cfg80211 snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq ee1004 joydev input_leds mei_me mei i915 snd_seq_device intel_pmc_core snd_timer pmt_telemetry snd soundcore pmt_class i2c_i801 intel_pch_thermal intel_vsec i2c_smbus i2c_mux serio_raw mac_hid drm_buddy acpi_pad sch_fq_codel nouveau mxm_wmi drm_gpuvm gpu_sched drm_ttm_helper ttm drm_exec drm_display_helper cec rc_core i2c_algo_bit msr dm_multipath parport_pc ppdev lp parport
[223527.007555] [      C6]  efi_pstore nfnetlink dmi_sysfs ip_tables x_tables autofs4 btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq raid1 raid0 linear hid_generic usbhid hid polyval_clmulni polyval_generic ghash_clmulni_intel sha256_ssse3 r8169 sha1_ssse3 psmouse intel_lpss_pci realtek ahci intel_lpss libahci idma64 video wmi pinctrl_sunrisepoint aesni_intel crypto_simd cryptd
[223527.010355] [      C6] CR2: fffffffffffffff8
```

Some of these messages seem to allude `8812au` (more specifically, the last call in the backtrace seems to be `rtw_sctx_done_err`), so I suspect the bug might have something to do with this.

Thank you very much for your help 🙏!

morrownr, 2025-09-13:
Hi @lgleznah 

From the README:

Notice: An updated standards compliant (mac80211), in-kernel driver for rtl8812au chipset based adapters and modules is available and as of kernel 6.14 is of good quality. If your distro uses kernel 6.14 or later, there is no need to install this driver. The in-kernel driver is part of the rtw88 series of drivers. The in-kernel driver is Linux Standards compliant (mac80211) and is a much better driver than this one. This driver will no longer get API related updates beyond kernel 6.14 (unless provided by a user). If you use a kernel prior to 6.14, it is possible to use the new standards compliant driver by going to the following repo:

https://github.com/lwfinger/rtw88

If you decide to use the in-kernel driver, remember to first remove the driver in this repo. You can run the following to remove the driver in this repo:

$ sudo sh remove-driver.sh

It has been my pleasure to maintain this driver for the last several years. Thanks to all of those who helped.

Remember: Ask not what your operating system can do for you, but what you can do for your operating system.

Regards,

@morrownr

cybersnow02, 2026-02-21:
@lgleznah did you find the issue to your problem? I ask this because your issue is probably not a driver related issue, nor a Kernel update issue. 3 things to check first: Try others USB ports, temporary remove any USB cord extension supplied with the 8812au unit and do a memory test with memtest to insure stability while the kernel is running. Also, you might try to update the driver with this new updated driver that support newer kernels (if you want to stay with original Realtek driver. Otherwise, you could try RTW88 driver that is included with Kernels >=6.14): https://github.com/cybersnow02/8812au-20210820