morrownr/8812au-20210820 #56: [Bug]: Error! Bad return status for module build on kernel: 6.16.7-arch1-1 (x86_64)

princeri, 2025-09-14:
### Checklist

- [x] I am not using this driver from inside a Virtual Machine.
- [x] I performed all of the [installation steps](/README.md#installation-steps) as written.
- [x] I have read the [FAQ](/FAQ.md) and my question is not answered there.

### Is Secure Boot enabled?

No

### uname

Linux archlinux 6.16.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 11 Sep 2025 17:42:36 +0000 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 04f2:b628 Chicony Electronics Co., Ltd EasyCamera Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU] Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub Bus 003 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) Bus 003 Device 004: ID 0cf3:e500 Qualcomm Atheros Communications 

### rfkill

0: ideapad_wlan: Wireless LAN 	Soft blocked: no 	Hard blocked: no 1: ideapad_bluetooth: Bluetooth 	Soft blocked: yes 	Hard blocked: no 2: hci0: Bluetooth 	Soft blocked: yes 	Hard blocked: no 3: hci1: Bluetooth 	Soft blocked: yes 	Hard blocked: no 4: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
rtl8812au/5.13.6-23: added
```

### iw

```text
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 74:fe:ce:d4:42:89
		ssid Ankur Gupta
		type managed
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### git status

```text
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```

### install-driver.sh

```text
: ---------------------------
: install-driver.sh v20241003
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 2/2 (in-use/total processing units)
: 3453752 (total system memory)
: 6.16.7-arch1-1 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 15.2.1 20250813
: dkms-3.2.1
: mokutil not installed (Secure Boot status unknown)
: ---------------------------

Checking for previously installed drivers...
Removing a driver that was added to dkms.
Deleting module rtl8812au/5.13.6-23 completely from the DKMS tree.
Removing 8812au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8812au-5.13.6-23
Finished checking for and uninstalling previously installed drivers.
: ---------------------------

Starting installation.
Copying 8812au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8812au-5.13.6-23
Deprecated feature: CLEAN (/usr/src/rtl8812au-5.13.6-23/dkms.conf)
Creating symlink /var/lib/dkms/rtl8812au/5.13.6-23/source -> /usr/src/rtl8812au-5.13.6-23
The driver was added to dkms successfully.
: ---------------------------

Deprecated feature: CLEAN (/usr/src/rtl8812au-5.13.6-23/dkms.conf)
Sign command: /usr/lib/modules/6.16.7-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)...(bad exit status: 2)
Failed command:
kernelver=6.16.7-arch1-1 ./dkms-make.sh

Error! Bad return status for module build on kernel: 6.16.7-arch1-1 (x86_64)
Consult /var/lib/dkms/rtl8812au/5.13.6-23/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

### What happened?

It just failed to install the driver everytime.

This is the output of /var/lib/dkms/rtl8812au/5.13.6-23/build/make.log:

DKMS (dkms-3.2.1) make.log for rtl8812au/5.13.6-23 for kernel 6.16.7-arch1-1 (x86_64)
Sun Sep 14 01:40:57 PM IST 2025

Building module(s)
# command: kernelver=6.16.7-arch1-1 ./dkms-make.sh
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.16.7-arch1-1/build M=/var/lib/dkms/rtl8812au/5.13.6-23/build  modules
make[1]: Entering directory '/usr/lib/modules/6.16.7-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/rtl8812au/5.13.6-23/build'
  CC [M]  core/rtw_cmd.o
  CC [M]  core/rtw_security.o
core/rtw_cmd.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.16.7-arch1-1/build/scripts/Makefile.build:287: core/rtw_cmd.o] Error 1
make[4]: *** Waiting for unfinished jobs....
core/rtw_security.c:17:10: fatal error: drv_types.h: No such file or directory
   17 | #include <drv_types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[4]: *** [/usr/lib/modules/6.16.7-arch1-1/build/scripts/Makefile.build:287: core/rtw_security.o] Error 1
make[3]: *** [/usr/lib/modules/6.16.7-arch1-1/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.16.7-arch1-1/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/rtl8812au/5.13.6-23/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.16.7-arch1-1/build'
make: *** [Makefile:2516: modules] Error 2

# exit code: 2
# elapsed time: 00:00:01

cybersnow02, 2025-09-21:
Hi @morrownr and @princeri 

I made a patch so this 8812au driver will work with Kernel 6.15 and 6.16. I don't like the fact that newer kernels won't be supported anymore for this driver project. The main problem with rtw88 (see https://github.com/lwfinger/rtw88/issues/111) is that we cannot adjust the txpower anymore (higher value). So I request that @morrownr continue to update this driver until they find out a solution. Anyway, I tested my patch and it works perfectly under Debian 13 Trixie with Backport kernel 6.16.

@morrownr : can you integrate this patch to main code?

[patch_kernel_6.16.patch](https://github.com/user-attachments/files/22454137/patch_kernel_6.16.patch)

cybersnow02, 2025-09-21:
Also, I renamed 'remove-driver.sh' to 'uninstall-driver.sh'. Take this into consideration if applying my patch.

princeri, 2025-09-22:
> Hi [@princeri](https://github.com/princeri)
> 
> This driver won't compile on kernel 6.16 because it should not need to compile on any distro using kernel 6.14 or later. Did you check to see if your adapter was working before you tried to install this driver?
> 
> A very solid rtl8812au standards compliant driver was added to the kernel with kernel 6.14. Try running the following:
> 
> $ sudo sh remove-driver.sh
> 
> After reboot, see what you have.

after uninstalling the driver it uses the 
readlink /sys/class/net/wlan0/device/driver                                                                                                                       

../../../../../../bus/usb/drivers/rtw88_8821au

this driver has problem where the packet inject doesn't works sometime....
also unable to find wifi networs when in monitor mode 

the 8812au was working absolutely fine....

cybersnow02, 2025-09-22:
About the txpower issue for driver rtw88 (issue : https://github.com/lwfinger/rtw88/issues/111), I did some test. It is easy for me to know if txpower works or not. An easy answer would be that without your 8812au, I couldn't answer to you because I did most of the patch work sit down in a forest, way far away from the access point. Without txpower adjusted to max, I couldn't connect at all.

After kernel 616 update, I switched to new rtw88 driver but I couldn't connect anymore. So I decided to look at the issue. I discovered that iw (or iwconfig) commands doesn't show the correct txpower at all. In fact, it was fluctuating each second. Then, I tried to force it with 'iw phy1 set sar_specs 0 0:xx' (any number) without success. It remained at 20dBm all the time. So the rtw88 enforce the regulation and cannot be modified. Now, I'm thinking that Linux Kernel shouldn't force this kind of regulation to be 20dBm. If this thing doesn't break my hardware, then it shouldn't be locked at all. Doesn't the hardware just refuse the value if not supported? The hardware is the only one to authorize, not the kernel.

princeri, 2025-09-24:
> Hi [@morrownr](https://github.com/morrownr) and [@princeri](https://github.com/princeri)
> 
> I made a patch so this 8812au driver will work with Kernel 6.15 and 6.16. I don't like the fact that newer kernels won't be supported anymore for this driver project. The main problem with rtw88 (see [lwfinger/rtw88#111](https://github.com/lwfinger/rtw88/issues/111)) is that we cannot adjust the txpower anymore (higher value). So I request that [@morrownr](https://github.com/morrownr) continue to update this driver until they find out a solution. Anyway, I tested my patch and it works perfectly under Debian 13 Trixie with Backport kernel 6.16.
> 
> [@morrownr](https://github.com/morrownr) : can you integrate this patch to main code?
> 
> [patch_kernel_6.16.patch](https://github.com/user-attachments/files/22454137/patch_kernel_6.16.patch)

Can you please share me your patched version 8812au driver.

morrownr, 2025-09-24:
@cybersnow02 

If you are interested, I can give you admin rights to this repo so that you can take over maintenance of this repo. This might be the best solution. I would ask you to consider your changes carefully as I have numerous Realtek WiFi 5 vendor drivers up here and it is best if they work in a consistent manner. An example of what I am talking about is the change from `remove` to `uninstall`. I understand the desire as I am the one that made that change starting with the Realtek WiFi 6 drivers... however I do not think it is a good change for WiFi 5 at this point as keeping things consistent would require the change in all drivers and in the docs and in the mind of all current users... and this is a repo that gets a lot of hits per day. I will leave things alone as far as API changes go as long as you don't break things but other things that can lead to a lot of message traffic is something to avoid. 

What do you think?

cybersnow02, 2025-09-24:
By the way, I just realized that I made the patches according to the following commit in February 21, 2025. So before applying my patches, download the right commit main code: https://github.com/morrownr/8812au-20210820/commit/dabcb74da670a12ca74125588bc89aa8ad086c13

At this stage, it already included the Kernel 6.14 patches. thanks